ArduCAM/MIPI_Camera #2: How is it that the hardware is out, but the software featured prominently on the advertising blog and video is not posted anywhere here on GitHub under the Arducam account?

ArduCAM, 2019-05-18:
How is it that the hardware is out, but the software featured prominently on the advertising blog and video is not posted anywhere here on GitHub under the Arducam account?
There are virtually no instructions for building and installing the driver, but I am experienced, so that much was a non-issue.  But no matter what why is there is no reference to preprocess.sh or loaddriver.sh?

Crucially, I don't have any /dev/video/* handles.

So why aren't these new products even listed on the main ArduCam website?
Why do the buried product pages for them have blank documentation and software pages.

Please just post the proper code, here on GitHub?

_Originally posted by @UberEclectic in https://github.com/ArduCAM/RaspberryPi/issues/7#issuecomment-493647122_

ArduCAM, 2019-05-18:
We are updating the driver from the kernel to the userland SDK now because of the GPL license issue.
The video and blog is the old version, we will try to update them very soon. The old V4L2 driver was removed from the github, if you need it please send us email.
The quick start instruction of new userland driver can be found from the README file in this repo.

Dlahacz, 2019-05-23:
Hi, is this userland SDK somehow compatible with pythons picamera? I'm trying Motorized focus example from https://github.com/ArduCAM/RaspberryPi/tree/master/Motorized_Focus_Camera and I'm getting
```
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.
```
Capture example from this repo is working.
Thanks