ArduCAM/MIPI_Camera #6: Impossible to open the camera

JTaveau, 2019-06-19:
Hello, 
I bought the IMX298 camera module few weeks ago and try to use it.
I installed the complete opencv git on my rapsberry pi zero w, then I downloaded the driver for the arducam mipi camera, following the instructions.
When I want to launch an example, I write ./preview it returns :  
Open camera... 
init camera status = -1

Could you help me please ?

ArduCAM, 2019-06-19:
Can you send email to admin@arducam.com ?

lalinlalote, 2019-07-18:
Hello! i have the same issue!

LarryDong, 2019-08-12:
The same, too.

JeffJassky, 2019-08-14:
I think it has to do with the i2c interface.

It's defaulting to `/dev/i2c-`0 or `/dev/i2c/0` where I believe it should be `/dev/i2c-0` or `/dev/i2c/0`

However, I'm not sure how to fix it.

LarryDong, 2019-08-15:
> 
> 
> The same, too.

I know the reason now. I am using a thrid-part module. This driver only support camera module from arducam itself. I am considering to buy an arducam now.

jhaliburton, 2019-11-03:
I also have this problem. I am using an ArduCAM product board UC-599 w/ the OV9281. I'm running an RPI 2B (board revision 22042). I have also observed that the device registers on i2c-0 as opposed to i2c-1. I noticed that the camera_i2c script did not contain a value for my board revision. I have sent emails to ArduCAM and will follow up if/when they respond.

JeffJassky, 2019-11-03:
@ArduCAM Lee - any update on this? We cannot get these camera modules to work.

jhaliburton, 2019-11-05:
Thank you to Bin from the arducam support team for solving my issue (see above). They were able to provide me with an updated libarducam_mipicamera.so file and an updated camera_i2c file. After doing an install with the new scripts the UC-599 board containing the OV9281 camera is operational on my RPI 2B (rev 22042). Bin indicated that the new files would be pushed to git shortly. If you have had problems with your camera on RPI 2 models then hopefully this will resolve your issue.

ArduCAM, 2019-11-05:
> @ArduCAM Lee - any update on this? We cannot get these camera modules to work. It has been 6 months.

Sorry I missed your message, what's the problem? Please visit our forum to post a question. https://www.arducam.com/forums
Thanks.

ErHimani, 2022-03-31:
> Thank you to Bin from the arducam support team for solving my issue (see above). They were able to provide me with an updated libarducam_mipicamera.so file and an updated camera_i2c file. After [...]

sir can you tell how we get these updated files?