ArduCAM/MIPI_Camera #9: Exposure failure with ov9281_external_trigger.c

kpmoorse, 2019-08-20:
I am trying to run a handful of ArduCam's OV9281 modules with a synchronized external trigger, but I am running into issues when I run the ov9281_external_trigger.c code to change the trigger mode.

At the moment the write_regs() function is called (tested by changing the argument of the preceding usleep() function), the camera exposure seems to drop to nearly zero, outputting all black except when pointed at a bright light source. The below images are taken of my ceiling, before and after the change.

Reducing the clock pulse frequency (supplied via pigpio on a Raspberry Pi separate from the one controlling the camera) from 60Hz to 30Hz returns the camera to normal exposure but results in alternating frames of normal video and all black--presumably because the software is expecting twice as many frames as it is receiving.

My best guess is that at 60Hz, the camera's exposure isn't complete when the next trigger pulse comes in, leading to the described failure, but manually changing V4L2_CID_EXPOSURE and V4L2_CID_GAIN has no effect, regardless of the status of arducam_software_auto_exposure() and arducam_set_resolution(). Is there an undocumented loss of user control over exposure time associated with the register changes in write_regs()?

Other diagnostics notes:
- I have tried swapping out both the OV9281 and the Raspberry Pi it is connected to with no change
- I have checked the clock signal with an oscilloscope and it is a clean 3.3V pulse at the expected frequency

[image]
[image]

ArduCAM, 2019-08-28:
Please send me your trigger generator code as well. We will double check it tomorrow. Thanks

kpmoorse, 2019-09-05:
After further communication with Lee, I received updated code which does not produce the jitter effect. Repeated tests over the last few days confirm that the problem does not return intermittently.

closed 2019-09-05 by kpmoorse