ArduCAM/MIPI_Camera #11: Exposure (Focus) & white balance (brightness) Adjustment for Arducam 16 MP cam module

HarisFateh, 2019-10-03:
How to set the manual values for exposure or focus & white balance or brightness. It seems that if we do not adjust the camera in appropriate position, it doesn't provide the image in any better than poor quality image. I have tried multiple times to set the index and focus_val attribute but it doesn't help. Kindly make an Application based on python for Raspbian platform or at least provide the enough information for description to edit the focus & brightness value on hand. Because there  is no way to adjust the parameters while photography session.

nvki, 2019-10-25:
Hey, I have the same sensor - the readme in the rpi part of this repo seemed quite helpful to get started. There are a few python examples in the repo as well, see if they work to test if everything is set up right.
Then you can use the camera.set_control() function to control any value listed when running the list_format executable.

In python, after creating a camera instance and initiating it like in the examples you can:

camera.set_control(v4l2.V4L2_CID_EXPOSURE, 3000)

Since my sensor only arrived yesterday I cannot tell you what unit that represents, the max feels like 2-3 seconds

BlessSonic, 2019-11-23:
Hi, I'd appreciate some help on this. 

When executing the python capture I get only green pictures. [example](https://ibb.co/WkK01Mt)

Changing or removing the auto white balance line in the script does not have any effect on the picture. [capture.py:17](https://github.com/ArduCAM/MIPI_Camera/blob/87774fc0d1f77b79e1de7909ec1f10da9eb5743a/RPI/python/capture.py#L17)

I use RBP4 + 16 MP IMX298 

Thanks

nvki, 2019-11-25:
@BlessSonic: Did you try setting the digital gain registers as described [here](https://www.arducam.com/docs/camera-breakout-board/16mp-imx298/software-guide/)?

You can set the red/blue gain ( like you would with a kelvin temperature slider ) and the Gr/Gb gain ( like the 'tint' setting would do ). 

Eg. to tint the image more reddish, set Gr/Gb to 786 (default seems to be 1020):

camera.write_sensor_reg(0x020E, 3)
camera.write_sensor_reg(0x0214, 3)

all the assembling x bit numbers from 8 bit ones is a bit weird, but you'll get into it. Eg. digital_gain_Gr is a 12 bit value set via one 4 and one 8 bit register (0x020E and 0x020F): 

digital_gain_Gr = 0x020E(1-15) * 256 + 0x020F(0-255)
eg. with 0x02E = 7 and 0x020F = 16: 7*256 + 16 = 1808
the maximum value is 2^12 = 4096

to calculate the two values from a single input value in python you can use divmod:

hi, lo = divmod(val, 256)
camera.write_sensor_reg(args[0], hi)
camera.write_sensor_reg(args[1], lo)

hope this helps, I am still new to this
[image]
[image]

nvki, 2019-11-25:
And yes, the auto white balance seems to be broken with the IMX 298 - maybe the dynamic library hasnt been updated with its register values yet...

BlessSonic, 2019-11-25:
@nvki Thank you very much for the response - it was helpful to me. I was able to manually change the registers.

I don't know enough about this project yet to adjust the dynamic libraries my self. 

May be some one from @ArduCAM is able to provide more infos.

nvki, 2019-12-02:
@BlessSonic Unfortunately I dont think we can make any adjustment to the dynamic library, since it doesnt seem to be open source.

ArduCAM, 2019-12-04:
We now moved our support to our forum, please resent the questions to https://www.arducam.com/forums/ 
Thank you.

HarisFateh, 2019-12-05:
@nvki Hi!
Can you help me adjust the white balance (brightness) via registers values?
And also if you be so kind to help me that if it's there any GUI that you showed works with Arducam 16 MP (MIPI) cam module? I was working on it but couldn't find any answers to these specifics.
Thanks in anticipation.

nvki, 2019-12-05:
@HarisFateh I can send you a cleaned version of one of my python scripts - however theres no GUI for that.
I think there is arducam software that lets you change register values via a GUI - as @ArduCAM suggested you could ask for it in the [new support forum](https://www.arducam.com/forums/) or see if you find a link in the [docs](https://www.arducam.com/docs/).

BlessSonic, 2019-12-05:
@nvki, @HarisFateh  There seems to be nothing on the new support forum yet. 

> @HarisFateh I can send you a cleaned version of one of my python scripts - however theres no GUI for that.

I'd be interested in the script as well

HarisFateh, 2019-12-06:
@nvki Thank you very much. Kindly, send me the script to my email address at "haris_fateh@live.com". That's so nice of you.
I have contacted that arducam support multiple times about their source code & module wrapping for different functions that need serious attention but yet they are working real slow. 
I hope your script just work fines & solve the problem regarding white balance.

ArduCAM, 2019-12-09:
The GUI is only for Windows which uses our USB camera shield to work with these camera breakout.
For Raspberrypi MIPI camera driver, there is only command line examples.
We have update the MIPI camera driver with optimized default settings, please try it out and ask for any questions on our new forum.
Thanks.