ArduCAM/MIPI_Camera #13: Recording or streaming over the network at resolution higher than 1920x1080

mythsmith, 2019-10-17:
I increased the resolution in video2stdout.c and video.c and I receive:

Failed to start video encoding, probably due to resolution greater than 1920x1080 or video_state setting error.

Is it possible to stream/record anything bigger than this resolution?

SenexCrenshaw, 2019-12-07:
Same here. the previews (preview_setMode) work great and preview the higher resolution. It is the video callback function in the closed source lib that is causing the issue.

adufftpc, 2020-03-04:
Same here. Need a possibility to record video larger than FullHD as well.

UCTRONICS, 2020-03-04:
@adufftpc 
Due to the RPI 264 encoder does not support more than 1080P resolution, so, it is not impossible to
encoder video over 1080P now.