ArduCAM/MIPI_Camera #16: OV9281 and Raspberry pi 4

Ninic0c0, 2019-11-05:
Hi, 

HW : OV9281 and raspberry pi 4B 4Go

SW: fresh buster install + deps

Could you please some info about this issue :

> 
> pi@raspberrypi:/home/pi/MIPI_Camera/RPI/python $ python capture.py 
> Open camera...
> init_camera: Unexpected result.
> pi@raspberrypi:/home/pi/MIPI_Camera/RPI/python $ i2cdetect -y -r 0
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
> 60: 60 -- -- -- -- -- -- -- -- -- -- -- 6c -- -- -- 
> 70: 70 -- -- -- -- -- -- --  
> pi@raspberrypi:/home/pi/MIPI_Camera/RPI $ ./capture2opencv 
> Open camera...
> init camera status = -1
> pi@raspberrypi:/home/pi/MIPI_Camera/RPI $ ./ov9281_external_trigger 
> Open camera...
> init camera status = -1
> pi@raspberrypi:/home/pi/MIPI_Camera/RPI $ ./preview-camera0 
> Open camera...
> Found sensor ov9281 at address 60
> 

Could you please provide a solution ? Thx in advance.

Note: seems ok on raspberry pi 3B

JirkaN1, 2019-11-16:
I have the same problem, OV9281 camera on rpi 1 run OK, but rpi 4/4GB throw segmentation fault.
Probed with the same SD card image with buster on both Rpis and then with fresh install on rpi 4.
Hardware of rpi 4 seems be OK, the original rpi camera capturing without any problem.
Result of the i2cdetect is the same, 0x6c
Thank You for any help.

JirkaN1, 2019-11-18:
Bad rpi 4 board!
Probed with another board and the same SD card, the camera working without any problem with arducam and original rpi cam.
The bad board have infinity resistance on pin 12 - CAM_CLK, on another pins are some MegaOhms.

Ninic0c0, 2019-11-18:
Hi @JirkaN1 , have you find any solution about this trouble ? I'm mean do you think hardware modification can solve this issue ?

Ninic0c0, 2019-11-19:
@ArduCAM Is it possible to have some help ? Tested on 3 raspberry pi 4 (buster full updated) and same issue, so maybe the sensor is bogus or it is not compatible with rpi4...

JirkaN1, 2019-11-19:
Hi Ninic0c0,
I don't think, that the rpi can be repaired home, under microscope find I no any defect.
Board have I changed by the seller, but after two days keyboard knocking
cant't capture faster than cca 60fps, it seems, that run only one lane even if is set mode 5. Have the M12 version two lanes @Arducam ?

Ninic0c0, 2019-11-20:
Hi @JirkaN1 & @ArduCAM ,

More info today after running _camera_i2c_ tools
pi@raspberrypi:~/MIPI_Camera/RPI $ ./preview-camera0 
Open camera...
Found sensor ov9281 at address 60
Setting the resolution...
Can't open the file
mmal: Failed to fix lens shading, use the default mode!
Current resolution is 1280x800
Notice:You can use the list_format sample program to see the resolution and control supported by the camera.
Start preview...
Reset the focus...
Failed to set focus, the camera may not support this control.
Setting the exposure...
Setting the exposure...
Setting the hfilp...
Enable Auto Exposure...
Enable Auto White Balance...
Mono camera does not support automatic white balance.
Setting the resolution...
Can't open the file
mmal: Failed to fix lens shading, use the default mode!
Failed to write register 
Current resolution is 2560x800
Stop preview...
Close camera...

JirkaN1, 2019-11-20:
It seems, that your camera and rpi is OK. 
If are You logged to rpi on ssh, the preview isn't visible, but is projected on hdmi output. This is normal state.

Ninic0c0, 2019-11-20:
I have already an HDMI display connected and no camera output...

JirkaN1, 2019-11-20:
video10 video11 video12
Try capture a image using capture_raw, the file must have 1024000Bytes.

glddiv, 2019-11-22:
Hi @Ninic0c0 
Sorry for replying so late.

> More info today after running camera_i2c tools
pi@raspberrypi:~/MIPI_Camera/RPI $ ./preview-camera0

Can you show me the output of the camera_i2c script?

glddiv, 2019-11-22:
Hi @JirkaN1 
Sorry for the late reply.

> cant't capture faster than cca 60fps, it seems, that run only one lane even if is set mode 5. Have the M12 version two lanes @ArduCAM ?

Can you tell me how you tested it?
Try using `./arducamstill -t 100000 -m 5 -awb 0 -ae 0` , which will display the correct frame rate
[image]

JirkaN1, 2019-11-22:
Hi @glddiv ,
Yes, I have tested the arducamstill yesterday, it use a raw_callback function for fps test and give me the the same result 145fps. (in sensor datasheet is 120fps as maximum)
But if I make a fast loop with use arducam_capture repeatedly, acquire raws and save them to ramdrive (or drop to NULL), I got exactly half of the number, 72fps. 
This seems, that the sensor capture every second frame. Is here any way to use raw_callback to save raw images, maybe in one file (without sensor stop) aka infinity analog film reel? It can be cutted later.  
If I capture 256 8bit frames, convert them to 16bit and then average them to one picture, got the true 16bit image almost without noise. But it must be done as fastest possible, because is possible the scene change in time. 
Thank You!

Ninic0c0, 2019-11-22:
Hi @glddiv ,

I'm not a work today but have made some modifications in the script : https://github.com/ArduCAM/MIPI_Camera/pull/15

Cheers,

glddiv, 2019-11-22:
Hi @Ninic0c0 ,
Thanks, it looks like a new Raspberry Pi 4 ,I didn't see it in the [Raspberry Pi revision codes](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md), I will check it out.

glddiv, 2019-11-22:
@JirkaN1 
arducam_capture is designed to get the latest frame and it doesn't read the image from the cache, so if you call arducam_capture not fast enough, you will lose some frames.
 If you don't want this effect, you can use [raw_callback](https://github.com/ArduCAM/MIPI_Camera/blob/master/RPI/raw_callback.c), but if you don't handle it fast enough in the callback, it will affect the overall speed.

JirkaN1, 2019-11-22:
Thank You @glddiv , 
Tested. 
The raw_callback makes always only 2-8 frames and then was probably some under/overrun occured.
In the function was the data transfer from buffer to (really big) cache only {pzasobnik = buffer->data;}
after function call I put sleep 4 seconds and then save data from cache to one file.
If I add the sleep to function, the cache was filled, but all frames are black. 
This fast acquisition isn't probably possible, I have some suspicion  to global shutter feature in sensor, rolling shutter acquiring data line by line and (in my opinion) gives the data more continuosly. Have @ArduCAM planned some more (5MP+) megapixel monochrome camera? 
TNX!

glddiv, 2019-11-25:
@glddiv 
After the callback function ends, the buffer will be freed for internal use. If you want to get data here you need to use a memory copy.

glddiv, 2019-11-25:
@JirkaN1 
After the callback function ends, the buffer will be freed for internal use. If you want to get data here you need to use a memory copy.

JirkaN1, 2019-11-25:
@glddiv 
Clearly :-) 
In the callback I make copy of buffer to uint8_t [1024000] array only and then close the callback. After it, make another [1024000] array, but uint16_t. Then I use for() loop to copy address by address from uint8 to uint16 array (raspberry make it in milliseconds), then make even more 255 exposures of the same and result of each exposure add to uint16 array. 
Voila... I have immediately 16bit grayscale raw without any delay by any additional computing. :-D 

Raspberry is amazingly power computer, if You use it for computing and not for clicking nonsenses. :-)

Ninic0c0, 2019-11-25:
Hello @glddiv ,

For info(after added c03112 as valid board):

`
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ ./camera_i2c
setting GPIO for board revsion: c03112
Raspberry Pi3B / Pi3B+ / 3A / 4B(1G/2G/4G)

Set state of 133 to 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- 6c -- -- -- 
70: 70 -- -- -- -- -- -- --                         
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ 
`

Cheers

glddiv, 2019-11-25:
@Ninic0c0 
The dynamic library has been updated, can you give it a try and tell me the result?

Ninic0c0, 2019-11-25:
@glddiv 
 it has been a long way with many toubles but after this commit b685d398fe5ac52723490e6ea465eb45d1841d08
i'm finally able to grab an image :) thx !

closed 2019-11-25 by Ninic0c0

glddiv, 2019-11-26:
@Ninic0c0 
@JirkaN1 
Great to hear you have solved your problems.
Let me know if you need more help.

delgoodie, 2020-01-04:
I have a similar problem, I am using two OV9281 arducam cameras (UC-580) combine with the compute module. When I use arducam's capture method I can get both pictures but I want to write to the registers of the cameras. Because the cameras are on the same address, and the compute module 3+ supports two i2c busses, I moved the second camera to dev/i2c-1. Prior to moving the bus, I was able to write to each cameras' registers but whenever I wrote to camera 1 it would make everything default on camera 0 and vice versa. Now that I have moved cam 1 to bus 1, I can read and write perfectly to camera 1 but can't write or read from camera 0. I still get two pictures, however! Any help would be much appriciated, ps: I am using the StereoPi board to breakout the compute module.