ArduCAM/MIPI_Camera #20: Arducam IMX 298 on Jetson Nano

julayelei, 2019-12-19:
Hello team of ArduCAM!
Today arrived my Arducam IMX 298 (https://www.amazon.com/Arducam-Camera-IMX298-Raspberry-Plugged/dp/B07W6LTFZC) but I'm trying to use it on a Nvidia Jetson Nano through CSI port but apparently it can't be recognized. I read that is compatible with RPi but I wanted to try it on the Jetson Nano because I'm doing some Deep Learning research and I've used other CSI cameras but none of them have the quality that this Arducam IMX 298 has and I really wanted to try it.
Is it possible to make it work on my Jetson Nano?
Thanks in advance!

ArduCAM, 2019-12-20:
@julayelei
The IMX298 now is only supported on RPI, we didn't have the camera driver on Jetson Nano. 
If you do like to use this camera on Jetson nano, you have to ues together with our USB3 camera shield + MIPI camera adapter board through USB3 port.

julayelei, 2019-12-20:
@ArduCAM OK! I'll give it a try.
Thanks!

Nuri-Aslan, 2020-01-02:
Do you have a plan to prepare a driver for Jetson Nano?

closed 2020-01-28 by julayelei