ArduCAM/MIPI_Camera #22: Preview video from Stereo Camera Hat is really DARK

openedhardware, 2020-02-14:
I am using RPi4 & Stereo Camera Hat.

I've tried `./arducamstill -t 0 -m 10 -awb 1 -ae 1`, but the preview video is pretty dark:
[image]
If I increase the brightness, it turns yellow...
Anything I can do to get normal video feed?

openedhardware, 2020-02-14:
Here is my current setup:
[image]

ArduCAM, 2020-02-15:
@openedhardware 
The arducamstill is using our MIPI camera driver which is receiving RAW data from the camera without any ISP processing, you can directly access the sensor exposure register to increase the exposure speed.
Or you can try to use the raspistill command which using Raspberry pi official camera driver to try it out.

openedhardware, 2020-02-15:
@ArduCAM 
Hmm, so can I use `raspistill` command to use the official camera driver?
Here is the output:
```
pi@raspberrypi:~ $ raspistill -o 1.jpg
Camera control callback  cmd=0x4f525245mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board
^Cmmal: Aborting program
```
What do you think?

ArduCAM, 2020-02-16:
Have you done a cold reboot for the hardware?
After run the arducamstill the board register might be changed, so you need to power down to restore the default settings for raspistill command

openedhardware, 2020-02-20:
@ArduCAM 
Yeah, of course. I had rebooted several times.

ArduCAM, 2020-02-24:
can you setup a remote TeamViewer for debugging?

openedhardware, 2020-02-27:
@ArduCAM 
Please let me know what I have to do for debugging..

closed 2020-03-31 by openedhardware

arnolde, 2022-01-15:
@openedhardware Could you please let us know how you solved this? I have the same problem currently. Thanks!