ArduCAM/MIPI_Camera #24: OV9281 occasional capture timeout on CM3

lukedempsey, 2020-03-13:
I am using a CM3 which interfaces with the OV9281 using a custom carrier board.

I have a python script based heavily off of the [capture2opencv.py](https://github.com/ArduCAM/MIPI_Camera/blob/master/RPI/python/capture2opencv.py) demo - it sits in a while loop capturing images. 

I have set my gpu allocation as following:
```
pi@raspberrypi:~ $ vcgencmd get_mem gpu
gpu=128M
```

My wiring is a little bit different to a standard CM3 carrier so I have reflected this change in my setup:
```
self._camera = arducam.mipi_camera()

# define hw wiring for OV9281
self.camera_interface = arducam.CAMERA_INTERFACE()
self.camera_interface.i2c_bus = 0
self.camera_interface.camera_num = 1
self.camera_interface.sda_pins = (28, 0)
self.camera_interface.led_pins = (44, 5)
self.camera_interface.shutdown_pins = (44,5)

self._camera.init_camera2(self.camera_interface)
self._fmt = self._camera.set_resolution(640, 400)
self._camera.software_auto_exposure(enable = True)
```

My main loop is pretty much identical to capture2opencv.py:
```
 while True:
            # update frame
            frame = self._camera.capture(encoding = 'i420')
            height = int(self.align_up(fmt['height'], 16))
            width = int(self.align_up(fmt['width'], 32))
            image = frame.as_array.reshape(int(height * 1.5), width)
            image = cv2.cvtColor(image, cv2.COLOR_YUV2BGR_I420)
```

My problem is that after 50 or so images, the arducam_capture() method will timeout and return a NoneType. ([Definition located here](https://github.com/ArduCAM/MIPI_Camera/blob/e99e80f680e895e72eb6ae82c3bd87e63745bb4a/RPI/python/arducam_mipicamera.py#L380))

My current work around is the very hacky following to catch the buffer being a Null Pointer. It will just reinitialise the camera again for another ~50 images before the error occurs again.
```
try:
    image = frame.as_array.reshape(int(height * 1.5), width)
except ValueError as e:
    print("ValueError: {}".format(e))
    self._camera.close_camera()
    self._camera.init_camera2(self.camera_interface)
    continue                                         
```

I have been using this camera with a 3B+ for several weeks and haven't had this problem.

ArduCAM, 2020-03-14:
@lukedempsey Would you please post the question on our forum : https://www.arducam.com/forum/

lukedempsey, 2020-03-25:
@ArduCAM , I posted [here](https://www.arducam.com/forums/topic/ov9281-occasional-capture-timeout-on-cm3/) last week. This has got me stumped. Any help would be appreciated.

lukedempsey, 2020-03-26:
Didn't explicitly find the root cause, but it was definitely a pebcak. Resolved in the above Arducam forum thread.

lukedempsey, 2020-04-09:
Was mistaken, the issue was intermittent enough that it convinced me that it was fixed.

Didn't find the root cause, however a quicker version of my workaround is to reset the resolution, rather than closing and opening the camera again.

lukedempsey, 2020-06-14:
> however a quicker version of my workaround is to reset the resolution, rather than closing and opening the camera again.

Note to anyone who tries this, I found this wasn't a great way to do it. It would cause the pi to get locked up to a point where it appeared that had shut down

UCTRONICS, 2020-06-15:
Hello, 
Are you sure you can only get 50 pictures each time?