ArduCAM/MIPI_Camera #25: Errors in arducam_mipicamera.py

fuzzybear62, 2020-03-24:
Hello,
in the MIPI_Camera/RPI/python/arducam_mipicamera.py

line 182:184
arducam_stop_preview = camera_lib.arducam_stop_preview
arducam_start_preview.argtypes = [c_void_p]
arducam_start_preview.restype = c_int

should be:
arducam_stop_preview = camera_lib.arducam_stop_preview
arducam_stop_preview.argtypes = [c_void_p]
arducam_stop_preview.restype = c_int

It seems that are missing the functions to for the rgainvalue and bgainvalue
arducam_get_gain
arducam_manual_set_awb_compensation

UCTRONICS, 2020-03-25:
Sorry to hear your problems. Don't worry and we will update our lib as soon as possible.

minewilliam, 2022-11-15:
> Sorry to hear your problems. Don't worry and we will update our lib as soon as possible.

[image]

minewilliam, 2024-12-12:
@UCTRONICS
[image]