ArduCAM/MIPI_Camera #28: Support for OV9281 on Jetson Nano

gituuu, 2020-04-13:
I bought an OV9281 for evaluation using Jetson Nano. I checked the jetvariety repo and discovered that still there is no support in the device tree.

**Please could you give me the relevant section of the device tree for the Jetson Nano and OV9281?**

I can build my own device tree but you would save me a *lot* of trouble if you could just give me the .DTS

Thank you.

ArduCAM, 2020-04-14:
The OV9281 camera you got is only for Raspberry pi. 
We now moved to another solution for Jetson Nano camera with different hardware solution.

gituuu, 2020-04-14:
Since I spent $100 buying the camera, I am going to try to get it working. There is no reason why it should not work, right?

If I can't get it working, what global shutter cameras do you have for the Jetson Nano?

gituuu, 2020-04-14:
Why do you list the OV9281 on your "Jetson Nano supported cameras" webpage?

https://www.arducam.com/product-category/nvidia-jetson-nano-camera-arducam/

?

ArduCAM, 2020-04-15:
it is different camera actually another adapter board on the bottom, it requires additional hardware to work with Jetson Nano. Please contact our sales for a upgrade solution.
Thanks.