ArduCAM/MIPI_Camera #40: arducam B0174 make error

bogaskilop, 2020-05-13:
i 'm trying to install arcudam B0174  but having some issues    and dont have  live prieview mode  arrducamstill can make a photo  and save it to jpg file but i do not have any live preview 

pi@raspberrypi:~/MIPI_Camera/RPI $ make clean && make
rm -f *.o
rm -f video4cameras preview_setMode arducamstill capture video list_format capture_raw raw_callback yuv_callback read_write_sensor_reg ov9281_external_trigger 2MPGlobalShuterDemo preview-camera0 preview-dualcam capture-dualcam video2stdout capture2opencv qrcode_detection  opencvGui  
gcc  -I. -g -O0 -std=gnu11 -o video4cameras video4cameras.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o preview_setMode preview_setMode.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o arducamstill arducamstill.c -larducam_mipicamera -lpthread
arducamstill.c: In function ‘main’:
**arducamstill.c:408:50: warning: passing argument 3 of ‘pthread_create’ from incompatible pointer type [-Wincompatible-pointer-types]
   int ret = pthread_create(&processCmd_pt, NULL, prcessCmd,&processData);
                                                  ^~~~~~~~~**
In file included from arducamstill.c:9:
/usr/include/pthread.h:236:15: note: expected ‘void * (*)(void *)’ but argument is of type ‘void (*)(PROCESS_STRUCT *)’ {aka ‘void (*)(struct <anonymous> *)’}
       void *(*__start_routine) (void *),
       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
gcc  -I. -g -O0 -std=gnu11 -o capture capture.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o video video.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o list_format list_format.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o capture_raw capture_raw.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o raw_callback raw_callback.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o yuv_callback yuv_callback.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o read_write_sensor_reg read_write_sensor_reg.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o ov9281_external_trigger ov9281_external_trigger.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o 2MPGlobalShuterDemo 2MPGlobalShuterDemo.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o preview-camera0 preview-camera0.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o preview-dualcam preview-dualcam.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o capture-dualcam capture-dualcam.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o video2stdout video2stdout.c -larducam_mipicamera -lpthread
g++ -I. -g -std=gnu++11 `pkg-config --cflags --libs opencv` -o capture2opencv capture2opencv.cpp -larducam_mipicamera -lpthread
g++ -I. -g -std=gnu++11 `pkg-config --cflags --libs opencv` -o qrcode_detection qrcode_detection.cpp -larducam_mipicamera -lpthread -lzbar
g++ -I. -g -std=gnu++11 `pkg-config --cflags --libs opencv` -o opencvGui opencvGui.cpp -larducam_mipicamera -lpthread

so i tried  with that option  **[-Wincompatible-pointer-types]** and the result is :

pi@raspberrypi:~/MIPI_Camera/RPI $ make clean && make -Wincompatible-pointer-types
rm -f *.o
rm -f video4cameras preview_setMode arducamstill capture video list_format capture_raw raw_callback yuv_callback read_write_sensor_reg ov9281_external_trigger 2MPGlobalShuterDemo preview-camera0 preview-dualcam capture-dualcam video2stdout capture2opencv qrcode_detection  opencvGui
gcc  -I. -g -O0 -std=gnu11 -o video4cameras video4cameras.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o preview_setMode preview_setMode.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o arducamstill arducamstill.c -larducam_mipicamera -lpthread
arducamstill.c: In function ‘main’:
arducamstill.c:426:50: warning: passing argument 3 of ‘pthread_create’ from incompatible pointer type [-Wincompatible-pointer-types]
   int ret = pthread_create(&processCmd_pt, NULL, prcessCmd,&processData);
                                                  ^~~~~~~~~
In file included from arducamstill.c:9:
/usr/include/pthread.h:236:15: note: expected ‘void * (*)(void *)’ but argument is of type ‘void (*)(PROCESS_STRUCT *)’ {aka ‘void (*)(struct <anonymous> *)’}
       void *(*__start_routine) (void *),
       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
gcc  -I. -g -O0 -std=gnu11 -o capture capture.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o video video.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o list_format list_format.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o capture_raw capture_raw.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o raw_callback raw_callback.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o yuv_callback yuv_callback.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o read_write_sensor_reg read_write_sensor_reg.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o ov9281_external_trigger ov9281_external_trigger.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o 2MPGlobalShuterDemo 2MPGlobalShuterDemo.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o preview-camera0 preview-camera0.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o preview-dualcam preview-dualcam.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o capture-dualcam capture-dualcam.c -larducam_mipicamera -lpthread
gcc  -I. -g -O0 -std=gnu11 -o video2stdout video2stdout.c -larducam_mipicamera -lpthread
g++ -I. -g -std=gnu++11 `pkg-config --cflags --libs opencv` -o capture2opencv capture2opencv.cpp -larducam_mipicamera -lpthread
g++ -I. -g -std=gnu++11 `pkg-config --cflags --libs opencv` -o qrcode_detection qrcode_detection.cpp -larducam_mipicamera -lpthread -lzbar
g++ -I. -g -std=gnu++11 `pkg-config --cflags --libs opencv` -o opencvGui opencvGui.cpp -larducam_mipicamera -lpthread

please advice  as for now  this camera is just a electronic trash if software can not be compiled on fresh raspbian on RPI 4.

bogaskilop, 2020-05-14:
really no one can tell what to do with that issue ?  
becosue of that when i run arducamstill i do not see any live preview from camera 

arducamstill.c: In function ‘main’:
arducamstill.c:408:50: warning: passing argument 3 of ‘pthread_create’ from incompatible pointer type [-Wincompatible-pointer-types]
int ret = pthread_create(&processCmd_pt, NULL, prcessCmd,&processData);
^~~~~~~~~
In file included from arducamstill.c:9:
/usr/include/pthread.h:236:15: note: expected ‘void * ()(void )’ but argument is of type ‘void ()(PROCESS_STRUCT )’ {aka ‘void ()(struct *)’}
void (__start_routine) (void *),
~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~

ArduCAM, 2020-05-21:
Don't worry. Please attach me the messages when the arducamstill is running.

bogaskilop, 2020-06-22:
ok  today i will do it  as i got only screen shots  now but here there is no option to upload screen shot 
but  in console when  ardustill run in console  i goes  till  green 
Strat preview....
and under this is  green fonts  with camera parameters  and  no pop up window  with preview  form camera