ArduCAM/MIPI_Camera #41: Cannot install Mipi driver on Jetson Nano with JetPack 4.4

sputnick, 2020-05-19:
The instructions to get going with the Jetvariety camera boards (UC-599) doesn't work on a brand new Jetson Nano with the latest downloaded SD card image. See below. 

$ sudo dpkg -i ./arducam-nvidia-l4t-kernel_4.9.140-32.3.1-20200325151719_arm64.deb
dpkg: regarding .../arducam-nvidia-l4t-kernel_4.9.140-32.3.1-20200325151719_arm64.deb containing arducam-nvidia-l4t-kernel, pre-dependency problem:
 arducam-nvidia-l4t-kernel pre-depends on nvidia-l4t-ccp-t210ref
  nvidia-l4t-ccp-t210ref is not installed.

dpkg: error processing archive ./arducam-nvidia-l4t-kernel_4.9.140-32.3.1-20200325151719_arm64.deb (--install):
 pre-dependency problem - not installing arducam-nvidia-l4t-kernel
Errors were encountered while processing:
 ./arducam-nvidia-l4t-kernel_4.9.140-32.3.1-20200325151719_arm64.deb

$ uname -a
Linux nano 4.9.140-tegra #1 SMP PREEMPT Wed Apr 8 18:10:49 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
$ cat /etc/nv_tegra_release
# R32 (release), REVISION: 4.2, GCID: 20074772, BOARD: t210ref, EABI: aarch64, DATE: Thu Apr  9 01:22:12 UTC 2020

glddiv, 2020-05-19:
Hi @sputnick ,
Currently the deb installation package of Mipi driver on Jetson Nano only supports L4T32.3.1 which is JetPack 4.3. We are adapting L4T32.4.2, which will be completed in the next few days.

glddiv, 2020-05-20:
Hi @sputnick ,
We have updated the support of L4T32.4.2 in commit 462c76232d0881fa59d22f64aa8eb1f84ef39df9, please check it.

gokhankucukayan, 2020-05-20:
Hi @glddiv,
Firstly thanks for fast development. But there is a problem on new support. 

$ sudo dpkg -i arducam-nvidia-l4t-kernel_4.9.140-32.4.2-20200520093606_arm64.deb 
Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 146623 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.2-20200520093606_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.2-20200520093606) ...
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.2-20200428140713); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.2-20200408182156.

$ uname -a
Linux asis-nano 4.9.140-tegra #1 SMP PREEMPT Wed Apr 8 18:10:49 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux

$ cat /etc/nv_tegra_release 
# R32 (release), REVISION: 4.2, GCID: 20074772, BOARD: t210ref, EABI: aarch64, DATE: Thu Apr  9 01:22:12 UTC 2020

glddiv, 2020-05-21:
Hi @gokhankucukayan ,
It looks like nvidia has two versions of l4t32.4.2 sd card image, the version I installed is 4.9.140-tegra-32.4.2-20200428140713, I will confirm it.

gokhankucukayan, 2020-05-21:
Hi @glddiv again,
Sorry, there is no link. I installed through sdk-manager with usb boot cable. But in sdk manager my choice was "JetPack 4.4 DP" through target operating system.

glddiv, 2020-05-22:
@gokhankucukayan 
> Hi @glddiv again,
Sorry, there is no link. I installed through sdk-manager with usb boot cable. But in sdk manager my choice was "JetPack 4.4 DP" through target operating system.

I simply changed the dependency from nvidia-l4t-kernel (= 4.9.140-tegra-32.4.2-20200428140713) to nvidia-l4t-kernel (= 4.9.140-tegra-32.4.2-20200408182156), I have not Test it, if you want to try to install it, it is recommended that you make a data backup. After installing the deb package, the kernel and device tree files will be replaced.

sputnick, 2020-05-29:
New driver works for me on JetPack 4.4 thanks