ArduCAM/MIPI_Camera #45: parameter exposure

tiralonghipol, 2020-06-11:
hi

I wrote a ros wrapper for the arducam stereo hat ([link](https://github.com/tiralonghipol/raspicam_stereo)) and I was wondering if you could expose more parameters from the library. 
My particular interest is (if possible) the binning factor, in order to stream lower resolution images but higher fps.
The current implemented formats (1600x600, mode 7 if I am not wrong) still too big to have any real-time operation in ros / opencv

Any comment / help is highly appreciated !

P.

UCTRONICS, 2020-06-15:
Hello, which resolution do you want and I will try my best to help you configure it. What's more, Please tell me your detail operation steps on ros and I can test it with you.

tiralonghipol, 2020-06-15:
@UCTRONICS 
if you have a look here https://github.com/UbiquityRobotics/raspicam_node/blob/kinetic/cfg/Camera.cfg

I belive these are all the possible parameters one could take out of raspicam

Is something similar possible?

UCTRONICS, 2020-06-22:
Hello,
Sorry for missing your question. I have checked your Camera.cfg. We have release a little control parameters for users.  Due to we don't have ISP process, some control parameters we can't create.

tiralonghipol, 2020-06-22:
Sounds good, I will check it asap.
Can you provide some .c example with the updated parameters also?
Or a little description of what changed?
Thank you

tiralonghipol, 2020-07-02:
Hi again @UCTRONICS 
I figured out a way to set the parameters I want (almost) directly writing on the imx219 control registers
but a weird behavior happens now: I can correctly set, for example, the binning mode to a factor of x2 and the binning type as 'sum' instead of the standard 'average' , but for some reason the output of both left and right camera is 'zoomed' from the top left corner.
Here is an example of what I am talking about:
Original:
[image]

Binned (x2):

[image]

tiralonghipol, 2020-07-15:
@UCTRONICS 
any update on this?
also, can you add a mode for VGA resolution please?
Thank you in advance!