ArduCAM/MIPI_Camera #47: TypeError: VideoCapture() takes at most 1 argument (2 given)

razvanphp, 2020-06-26:
I'm trying to run the examples, after fixing the `TypeError: unsupported operand type(s) for +: 'range' and 'list'` error, but still no luck.

I'm using `Python 3.6.9` and `python3-opencv arm64 3.2.0+dfsg-4ubuntu0.1`

Any hints?

Side-question: can we use AR1820HS in gstreamer?

RustyRaptor, 2020-07-15:
What version of OpenCV are you using?

kskrueger, 2020-09-07:
This is an issue due to using OpenCV 3 (using apt install python3-opencv will get you OpenCV3). If you update to OpenCV4 it should fix the issue.

pinhaocheng, 2021-04-30:
Just in case anyone else is having the same issue, try [this solution mentioned here.](https://github.com/AlexJinlei/python-v4l2/issues/5#issuecomment-609050025)

> edit the file and wrap the `range(...) ` in a `list()`, e.g., `list(range(3, 4))`