ArduCAM/MIPI_Camera #49: Ethernet Link Down on OV9281 Cam Access on Raspi 4

stefanschuermans, 2020-06-28:
I'm using ArduCam OV9281 with Raspberry Pi 4 B 4GB. The camera is working fine. However, on first access to the camera, the wired ethernet interface loses its link. It is not possible to re-active the link by re-plugging the cable or "ifconfig eth0 up". After reboot, the ethernet link is working again - until the next camera access.

Please see the attached text files for details.

If there is any information missing or there is something I should try in my setup, please let me know.

[boot_config.txt](https://github.com/ArduCAM/MIPI_Camera/files/4842186/boot_config.txt)
[cam_access.txt](https://github.com/ArduCAM/MIPI_Camera/files/4842187/cam_access.txt)
[dmesg.txt](https://github.com/ArduCAM/MIPI_Camera/files/4842188/dmesg.txt)
[info.txt](https://github.com/ArduCAM/MIPI_Camera/files/4842189/info.txt)
[lsmod.txt](https://github.com/ArduCAM/MIPI_Camera/files/4842190/lsmod.txt)
[uname_a.txt](https://github.com/ArduCAM/MIPI_Camera/files/4842191/uname_a.txt)

ArduCAM, 2020-06-29:
Hello,
i have tested it and it can work normal. the firmware I am using is raspberrypi 4.19.118-v7l+ #21 SMP Thu Jun 11 
[image]
[image]

stefanschuermans, 2020-06-29:
Thank you for testing. I have tried to upgrade my firmware from `Linux raspi4 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux` to the latest one`Linux raspi4 5.4.49-v7l+ #1323 SMP Fri Jun 26 14:40:18 BST 2020 armv7l GNU/Linux`, but it did not help.

However, after some more testing of different versions, I have found out that the `wiringpi` version seems to be related to the issue. With `wiringpi-latest.deb` (version 2.52, from link in `README.md`), the Ethernet link goes down each time I access the camera. After downgrading to `wiringpi` version 2.50 (from Raspberry Pi OS APT repo), the Ethernet link stays up when using the camera.

Qengineering, 2020-07-24:
Sold an application with the OV9281 in external trigger mode. Measuring mushrooms on a belt and sending the diameter of the hood to a server via Ethernet. Need I say more? After an update with the latest wiringpi version Ethernet went down. Took me a full day debugging. In the end, check this repo. Many thanks Stefan Schuermans !! You saved my day and client !!