ArduCAM/MIPI_Camera #52: Jetvariety kernel driver for Xavier NX

razvanphp, 2020-07-15:
The current 4.2 driver does not work:
```
jetson@xavier:~$ sudo dpkg -i arducam-nvidia-l4t-kernel_4.9.140-32.4.2-20200520093606_arm64.deb
Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 173761 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.2-20200520093606_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.2-20200520093606) ...
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.2-20200428140713); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.3-20200625213407.

dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel
jetson@xavier:~$ uname -a
Linux xavier 4.9.140-tegra #1 SMP PREEMPT Thu Jun 25 21:22:12 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
jetson@xavier:~$ cat /etc/nv_tegra_release
# R32 (release), REVISION: 4.2, GCID: 20074772, BOARD: t186ref, EABI: aarch64, DATE: Thu Apr  9 01:26:40 UTC 2020
jetson@xavier:~$
```

Also, nvidia just released L4T revision 4.3, how can we get the drivers?

glddiv, 2020-07-16:
Hi @razvanphp 
The latest deb package has been updated in commit bf27ff677783f75de619c50cf349cffc2ebdce55, please check it.

razvanphp, 2020-07-16:
Thank you for the quick response, but does not work, maybe it is only for jetson nano? I'm testing on jetson xavier NX.

```
jetson@xavier:~$ sudo dpkg -i arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200716095813_arm64.deb
Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 141631 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200716095813_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200716095813) ...
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213809); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.3-20200625213407.

dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel
```

SMSmith, 2020-07-16:
I have the exact same problem as @razvanphp.

glddiv, 2020-07-17:
@razvanphp 
Yes, currently we only provide the driver for Jetson Nano, the driver for jetson xavier NX is under development, but currently we have encountered some resistance and it will take some time to solve the problem.

SMSmith, 2020-07-17:
@glddiv Can we help?  What needs to be done?

SMSmith, 2020-07-20:
@glddiv Is this days, weeks, or months away from being developed?  If you had to guess?

ArduCAM, 2020-07-20:
[
[Jetson_Xavier _NX_Driver.zip](https://github.com/ArduCAM/MIPI_Camera/files/4948767/Jetson_Xavier._NX_Driver.zip)
](url)
You can try the test version first.

SMSmith, 2020-07-23:
Is that version for L4t 32.4.2?  Are you working on a version for L4t 32.4.3?

razvanphp, 2020-07-24:
@SMSmith it's added here: https://github.com/ArduCAM/MIPI_Camera/blob/master/Jetson/Jetvariety/driver/XavierNx/arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200724104443_arm64.deb

razvanphp, 2020-07-24:
works, thank you!

```
jetson@xavier:~$ sudo dpkg -i arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200724104443_arm64.deb
Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 143016 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200724104443_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200724104443) ...
Setting up arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200724104443) ...
jetson@xavier:~$ v4l2-ctl --list-devices
vi-output, arducam-csi2 9-000c (platform:15c10000.vi:0):
	/dev/video0

jetson@xavier:~$ cat /etc/nv_tegra_release
# R32 (release), REVISION: 4.3, GCID: 21589087, BOARD: t186ref, EABI: aarch64, DATE: Fri Jun 26 04:34:27 UTC 2020
```

closed 2020-07-24 by razvanphp