ArduCAM/MIPI_Camera #53: Driver for Xavier NX doesn't work

cisvhat, 2020-07-23:
Hi, Thank you for your recent effort to launch camera driver for Jetson Xavier NX.
I'm trying to connect OV2311 camera by using arducam board to Xavier NX, but currently it's not working.

I cant find /dev/video* directory after I installed the driver and rebooted the system.
My kernel version is [ 4.9.140 ] and L4T release number is [ R32 revision 4.2 ] .
I'm not sure if this is related to driver or my mistake in the process of connecting camera.
If you had any suggestion and shared it , I would really appreciate for it.

ArduCAM, 2020-07-23:
Please check the driver below and make sure you are using the same kernel version (I think it is the same as yours)
https://github.com/ArduCAM/MIPI_Camera/tree/master/Jetson/Jetvariety/driver/XavierNx

cisvhat, 2020-07-23:
Thank you for reply.
I checked and the kernel version is same.

razvanphp, 2020-07-24:
worked for me with 4.3, you should upgrade anyway...

kskrueger, 2020-08-06:
@razvanphp What release of Jetpack are you running?