ArduCAM/MIPI_Camera #55: IMX477 FPS driver limit

razvanphp, 2020-07-24:
First of all thank you for compiling the driver for the sensor, much appreciated!

Originally when we bought the Arducam IMX477 cameras the description stated:
```
 Full resolution@60fps(normal), 4K2K@60fps, 1080p@240fps
```
As far as I understand, it is a limitation of the driver now, since the arducam version uses all 4 lanes and could handle the bandwidth.

Since the driver is open-source, can you please do the necessary changes to unlock the full potential of the camera and enable those modes too?

Thank you in advance!

glddiv, 2020-07-25:
Hi @razvanphp 
The biggest problem is that the csi interface of jetson nano or nx is 15pin, which only supports 2 lanes.

razvanphp, 2020-07-25:
then how does it work on jetvariety? it connects also on the CSI port. Any idea what the max FPS would be with 2 lanes?

razvanphp, 2020-09-20:
Any update on this?

We are testing with [JN30 auvidea carrier board](https://auvidea.eu/product/70787/) and this one has 4 lanes, can you please give us a test kernel version with full FPS or tell us what to modify?

Thank you!

dustinkerstein, 2021-03-08:
Was just curious, is there any update on this? Thanks!

rem0nz, 2021-05-27:
Me too, any updates on this please @ArduCAM ?

glddiv, 2021-06-07:
Hello everyone, 
sorry for the late reply, 
Here is the relevant response:
https://www.arducam.com/forums/topic/stereo-vision-with-auto-focus-hq-camera-at-12mp60fps/#post-71524

The current main problem is that there are too many versions of third-party carrier boards, and the device tree configuration of each carrier board may be different, and it is difficult for us to unify them.

I have a question, what do you need for 4k@60fps? From the technical manual, Jetson NX/Nano seems to be unable to encode at 4k@60fps.

If you have special needs, you can contact sales@arducam.com for customization.

rem0nz, 2021-06-07:
[https://tr.cloudmagic.com/h/v6/emailtag/tag/2.0/1623033247/c5785eb850efdabaaea7810de97b2769/2/c4a7fe39c3831da4359eaf8a32d701cf/6274a75b4dda249037176a520eebaaf3/9efab2399c7c560b34de477b9aa0a465/newton.gif]
My question was , the master /slave drive doesn’t really work on NX because it doesn’t support the latest JetPack.
Any updates on that release time ?

https://www.arducam.com/forums/topic/stereo-vision-with-auto-focus-hq-camera-at-12mp60fps/#post-71524<https://tr.cloudmagic.com/h/v6/link-track/1.0/1623033263571095-572a24cc-bd68-a00e-f52b-a3938255010c/1623033247/c5785eb850efdabaaea7810de97b2769/6274a75b4dda249037176a520eebaaf3/9efab2399c7c560b34de477b9aa0a465?redirect_uri=https://www.arducam.com/forums/topic/stereo-vision-with-auto-focus-hq-camera-at-12mp60fps/#post-71524>

I have a question, what do you need for ***@***.***? From the technical manual, Jetson NX/Nano seems to be unable to encode at ***@***.***

If you have special needs, you can contact ***@***.******@***.***> for customization.

glddiv, 2021-06-07:
Hi @rem0nz 
For other questions, please open a new post or contact us at support@arducam.com.
Short answer:
At present, there are some limitations in the master-slave mode of nx, so we have not released a new version of the driver.

Xrockyman, 2022-12-13:
> Any update on this?
> 
> We are testing with [JN30 auvidea carrier board](https://auvidea.eu/product/70787/) and this one has 4 lanes, can you please give us a test kernel version with full FPS or tell us what to modify?
> 
> Thank you!

Dear razvanphp,

Did you had success using imx477 and Auvidea JN30 carrier board ?

Thank you in advance