ArduCAM/MIPI_Camera #56: Will IMX477 driver work with Original RPi HQ camera also?

SaadTiwana, 2020-07-28:
Hi,

I see that you have shared a compiled version of the IMX477 driver from Ridgerun for the Jetson. Since it works for the Arducam IM477 sensor, would it also work for the original RPi HQ cam (v3)? Assuming the R8 removal modification is done as needed.

Thanks.

ArduCAM, 2020-07-28:
Yes, I think so as long as the hardware is functioning properly with R8 removed.

SaadTiwana, 2020-07-28:
Hi, thank you for your quick response. So I actually did try with the RPi Cam v3 with your driver on the Xavier NX (I made sure the versions were correct as stated in your documentation). However the camera wasn't detected. R8 had been removed and the board was electrically good (I verified). 
Is it possible to see the Adrucam IMX477 board schematics so I can compare if anything is different. RPi Cam V3 schematics are available.

ArduCAM, 2020-07-29:
Our engineer has confirmed that the the RPI Cam V3 is working by removing the R8 resistor with our Jetson Nano driver, there must be something wrong with your hardware.

SaadTiwana, 2020-07-30:
Indeed you're right. I finally found the issue. I was using the root On SSD to boot on my Jetson Xavier NX, as detailed in the following
https://github.com/jetsonhacks/rootOnNVMe
However, it turns out that the system actually boots the drivers from the SD Card before switching to the SSD. So whenever i would install the Arducam camera driver package it would copy the files to the SSD filesystem ,while the drivers are booted from the SSD. I simply switched to the SD card, installed the drivers and then it worked without problems. 

Thank you for your help and for making the driver available. It certainly saved a lot of time. Perhaps next time i can consider the Arducam camera instead.

closed 2020-07-30 by SaadTiwana