ArduCAM/MIPI_Camera #57: No camera available in jetson nano

fc73, 2020-07-30:
Hi,

I met a problem when I want to run IMX477 on a jetson nano.

First I downloaded the newest sd card image for nano and burn it into an empty sd card.

Then I download the driver of imx477 here for nano with the corresponding version.

When I want to run the command to pop up the camera, it showed 'no cameras available'.

Ps: my kernel version is a little bit strange, ending with '-tegra'.

It would be kind if someone can tell me how to fix those issues.

Thanks!

ArduCAM, 2020-07-31:
Would you please post a question on our forum with screenshot for each steps, our engineer will help you there.
https://www.arducam.com/forum/

fc73, 2020-08-02:
> Would you please post a question on our forum with screenshot for each steps, our engineer will help you there.
> https://www.arducam.com/forum/

Hi, thanks for the info. I’ve already posted in the forum. Could you please approve the post?

airgo-ai, 2020-08-09:
Same thig happened to me. No VideoX listed under /dev after the driver was installed per instructions on clean Nano SD image.

Abdob, 2020-09-02:
I had the same problem with the jetson nx, and the fix was to remove the R8 resistor from the camera. https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson#Download_the_JetPack_4.4