ArduCAM/MIPI_Camera #58: AR1820HS on Jetson Nano, little FPS issue

Andoku, 2020-07-30:
Camera: AR1820HS
Board: Jetson Nano

Hi, I am testing AR1820HS camera on Jetson Nano following this article: [camera-demonstration](https://www.arducam.com/docs/camera-for-jetson-nano/mipi-camera-modules-for-jetson-nano/camera-demonstration/)

I am runing example like this:
`python arducam_displayer.py --width 4912 --height 3684`

I am getting only 5-6 FPS but the spec says it can give:
> 4:3 – 18 Mp full resolution at 15 fps (MIPI-4L)

Where is the problem?

ArduCAM, 2020-07-31:
The Jetson Nano develop kit only support 2-lane MIPI interface, so the bandwidth is reduced by half. and 5-6fps is stable frame rate on Jetson, I am not sure if we raise the fps even faster it will cause any issue.

Andoku, 2020-07-31:
@ArduCAM 
What about Jetson NX? Will FPS be the same or faster?