ArduCAM/MIPI_Camera #68: Driver does not install anymore with new Jetson Xavier build

isildur7, 2020-10-09:
I recently bought a Jetson Xavier NX and Arducam IMX477 MIni Version, and was trying to set it up. However the patch does not install because my kernel version is ever so slightly different. I get the following error with dpkg:

`dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213407); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.3-20200924161919.
`
Can you please tell me how to fix this mismatch? Is it possible to update the patch?

job-1994, 2020-10-11:
I have the same issue on the Jetson Nano, would also like to know how to fix.

sebmorales, 2020-10-11:
I tired:
$ sudo dpkg --force-all -i ardu...

 and it gave me a warning about the kernel mismatch but looks like it is running.

glddiv, 2020-10-12:
Hey guys, 
Did you install the system through SD Card Image?
If not, you can try the method of sebmorales.

> I tired:
$ sudo dpkg --force-all -i ardu...
and it gave me a warning about the kernel mismatch but looks like it is running.

TPU2020, 2020-10-16:
Same problem here
`Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 154704 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200721164156_arm64_imx477.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200721164156) ...
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213407); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.3-20200924161919.

dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel
`
I've even tried $ sudo dpkg --force-all -i ardu...

`Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200721164156) ...
dpkg: arducam-nvidia-l4t-kernel: dependency problems, but configuring anyway as you requested:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213407); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.3-20200924161919.

Setting up arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200721164156) ...
`

glddiv, 2020-10-20:
Hi @TPU2020 

> I've even tried $ sudo dpkg --force-all -i ardu...
>
> `Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200721164156) ...
> dpkg: arducam-nvidia-l4t-kernel: dependency problems, but configuring [...]

I don't see the error, is it not installed successfully?
After the system restarts, you can use dmesg | grep imx477 to confirm whether the imx477 driver is installed successfully. If the above command outputs imx477 related results, it proves that the driver has been successfully installed.

TPU2020, 2020-10-20:
This is the output
`xavier@xavier-desktop:~$ dmesg | grep imx477
[    1.956282] imx477 9-001a: tegracam sensor driver:imx477_v2.0.6
[    1.979917] imx477 9-001a: imx477_board_setup: error during i2c read probe (-121)
[    1.979956] imx477 9-001a: board setup failed
[    1.980048] imx477: probe of 9-001a failed with error -121
[    1.980367] imx477 10-001a: tegracam sensor driver:imx477_v2.0.6
[    2.003678] imx477 10-001a: imx477_board_setup: error during i2c read probe (-121)
[    2.003751] imx477 10-001a: board setup failed
[    2.003831] imx477: probe of 10-001a failed with error -121
`

glddiv, 2020-10-20:
Hi @TPU2020 
It seems that the driver has been successfully installed, but it does not detect the camera, which is most likely caused by a loose cable or a wrong connection.

TPU2020, 2020-10-20:
I've just tried another brand new cable, but problem remains.
I've also tested the camera on RPi4 and it was working nicely.

Also Software Updater recognize this as an error with unmet dependencies and tries to remove this driver.
[image]
[image]

johanso-au, 2020-10-20:
Experiencing the same issue as above. Seems to be a dpkg requirements issue that is having a flow on impact. Install from a brand new (downloaded yesterday) Jetson NX image.

`nvidia@nvidia-camtest:~/Downloads$ sudo dpkg -i arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20201009160531_arm64.deb`
`Selecting previously unselected package arducam-nvidia-l4t-kernel.`
`(Reading database ... 142228 files and directories currently installed.)`
`Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20201009160531_arm64.deb ...`
`Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20201009160531) ...`
`dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:`
` arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213407); however:`
`  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.3-20200924161919.`
` `
`dpkg: error processing package arducam-nvidia-l4t-kernel (--install):`
` dependency problems - leaving unconfigured`
`Errors were encountered while processing:`
` arducam-nvidia-l4t-kernel`

Using --force-all appears to configure the package, however upon running the example:

`nvidia@nvidia-camtest:~/MIPI_Camera/Jetson/Jetvariety/example$ python arducam_displayer.py -d 1`
`Firmware Version: 3`
`Sensor ID: 0x2311`
`Serial Number: 0xAD6AFFFF` (note: Actual S/N edited)
`Gtk-Message: 14:33:22.738: Failed to load module "canberra-gtk-module"`

Working backwards from the GTK Module error, we get:

`nvidia@nvidia-camtest:~/MIPI_Camera/Jetson/Jetvariety/example$ sudo apt install libcanberra-gtk-module libcanberra-gtk3-module`
`Reading package lists... Done`
`Building dependency tree       `
`Reading state information... Done`
`libcanberra-gtk3-module is already the newest version (0.30-5ubuntu1).`
`libcanberra-gtk3-module set to manually installed.`
`You might want to run 'apt --fix-broken install' to correct these.`
`The following packages have unmet dependencies:`
` arducam-nvidia-l4t-kernel : Depends: nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213407) but 4.9.140-tegra-32.4.3-20200924161919 is to be installed`
` libcanberra-gtk-module : Depends: libcanberra-gtk0 (>= 0.2) but it is not going to be installed`
`E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).`

The required version differences between 32.4.3-20200625213407 and 32.4.3-20200924161919 seem to have flow on effects for the rest of the required packages.

johanso-au, 2020-10-20:
Also have the same "No Cameras Available" error, however the cameras can be viewed by opening /dev/video0 and /dev/video1 through VLC.

`nvidia@nvidia-camtest:~/Downloads$ gst-launch-1.0 nvarguscamerasrc sensor_id=0 !    'video/x-raw(memory:NVMM),width=1920, height=1080, framerate=60/1' ! nvvidconv ! nvoverlaysink`
`Setting pipeline to PAUSED ...`
`Pipeline is live and does not need PREROLL ...`
`Setting pipeline to PLAYING ...`
`New clock: GstSystemClock`
`Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:557 No cameras available`
`Got EOS from element "pipeline0".`
`Execution ended after 0:00:00.291614269`
`Setting pipeline to PAUSED ...`
`Setting pipeline to READY ...`
`Setting pipeline to NULL ...`
`Freeing pipeline ...`

glddiv, 2020-10-21:
Hi @johanso-au 

> Also have the same "No Cameras Available" error, however the cameras can be viewed by opening /dev/video0 and /dev/video1 through VLC.

Jetvariety series does not support nvarguscamerasrc, only imx219 and imx477 support nvarguscamerasrc.

glddiv, 2020-10-21:
@TPU2020 
Can you provide a hardware connection diagram? I can confirm that there is no problem with the driver.
By the way, are you using the official Raspberry Pi camera? If you want to use it on Jetson, you need to make some hardware changes.

glddiv, 2020-10-21:
It seems that force installation will prompt a dependency problem. I repackaged the deb package based on 20200924161919 dependency.
https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.1/20200924161919.zip

johanso-au, 2020-10-21:
> It seems that force installation will prompt a dependency problem. I repackaged the deb package based on 20200924161919 dependency.
> https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.1/20200924161919.zip

I can confirm that this repackage successfully installs on the latest Jetson NX image without issues. After the libcanberra-gtk modules are manually installed, the Jetvariety arducam_displayer.py example runs without any errors (although I'm getting a plain white screen, I think that's something on my end).

GTRedE, 2020-10-22:
> It seems that force installation will prompt a dependency problem. I repackaged the deb package based on 20200924161919 dependency.
> https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.1/20200924161919.zip

This does not work for the newest build for the Nano nor the build available on NVIDIA's site on October 3rd.

glddiv, 2020-10-26:
Hi @GTRedE
l4t32.4.4 has been updated
https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.1

glddiv, 2020-10-26:
Hi @johanso-au 
For NX, you need to manually modify these depths from 10 to 16:
https://github.com/ArduCAM/MIPI_Camera/blob/41523d0c7d57209b0d1c5c941b2e6d9a41514bfc/Jetson/Jetvariety/example/utils.py#L68

We are very sorry for this problem, it will be fixed soon.

isildur7, 2020-10-26:
> Hi @GTRedE
> l4t32.4.4 has been updated
> https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.1

Hi @glddiv thanks so much for updating. I am so sorry but there is still a kernel mismatch..

```
Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 146069 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201026143033_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.4-20201026143033) ...
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.4-20201016124427); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.4-20201016123640.

dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel
```
Thanks for all the work, I understand this must a bit frustrating.

johanso-au, 2020-10-27:
I have the same error, unfortunately. Jetson Nano

```
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201026143033_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.4-20201026143033) over (4.9.140-32.4.4-20201026143033) ...
ls: cannot access 'extlinux.conf_bak': No such file or directory
ls: cannot access 'Image.bak': No such file or directory
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.4-20201016123640); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.4-20201016124427.

dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel
```

glddiv, 2020-11-02:
Hi @johanso-au 
Sorry for the late reply, are you using the system installed by SDCard Image or NVIDIA SDK Manger?

glddiv, 2020-11-02:
Hi @johanso-au 
I believe you downloaded the driver of Jetson Xavier NX, not Nano.

This is the driver for Jetson Nano:
https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.1/arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201026143033_arm64.deb

razvanphp, 2020-11-13:
It seems there is a new version of kernel for NX already:
```
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.4-20201016123640); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.4-20201027211332.
```
```
jetson@xavier:~$ ./install_imx477.sh
Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.140-tegra-32.4.4-20201027211332
Jetson type:  Xavier NX (Developer Kit Version)
jetson@xavier:~$
```

metanurb21, 2021-01-05:
Any new development on this. Got a brand new Xavier and IMX477 today.
`Linux jetson-desktop 4.9.140-tegra #1 SMP PREEMPT Tue Oct 27 21:02:46 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux`
`# R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t186ref, EABI: aarch64, DATE: Fri Oct 16 19:37:08 UTC 2020`

installed:
arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201027211332_arm64_imx477.deb

Installation does not fail but no sign of camera anywhere?
`v4l2-ctl --device /dev/video0`
`Failed to open /dev/video0: No such device or address`

`dmesg | grep imx47` lists no results.

Strange thing is, `dmesg | grep imx` gives:
`[    1.961721] imx219 9-0010: tegracam sensor driver:imx219_v2.0.6
[    1.985370] imx219 9-0010: imx219_board_setup: error during i2c read probe (-121)
[    1.986076] imx219 9-0010: board setup failed
[    1.986359] imx219: probe of 9-0010 failed with error -121
[    1.986834] imx219 10-0010: tegracam sensor driver:imx219_v2.0.6
[    2.010133] imx219 10-0010: imx219_board_setup: error during i2c read probe (-121)
[    2.010437] imx219 10-0010: board setup failed
[    2.010675] imx219: probe of 10-0010 failed with error -121`

I don't think I installed this.

Any help would be much appreciated.

galAcarteav, 2021-03-16:
someone got any success on Nano with jet pack 4.5.1?