ArduCAM/MIPI_Camera #69: terminal freeze after running arducamstill

oiluigio, 2020-10-16:
Hi there,

I am trying to get a MIPI_camera (IMX298 16MP) working on a RaspberryPi 4.
My distro is Linux raspberrypi 4.19.108-v7l+ #1298 SMP Fri Mar 6 18:09:54 GMT 2020 armv7l GNU/Linux

I followed the README.md of MIPI_Camera/RPI in this repository to prepare dependencies and compile all the .c
Everything looked fine. Then I try  ./arducamstill -? and the output is correct

[image]

but the terminal appears freezed in the end.
Even worse the ssh connection to the pi is compromised and I need to unplug the device to regain control as I am using an headless version. It looks like everything get freezed after the script end.

I also tried to take picture ./arducamstill -o test.jpg and the picture is there, but again everything freeze.

The same is happening with all the scripts I tested  (i.e. ./capture and /list_format)

I tried to add & in the end of the line or to print the log elsewhere but it always ends the same way.

Hope you can help!

ArduCAM, 2020-10-20:
Hello,
It seems the system is too old. Some customers encountered similar problems before. Please try to use the new release firmware.
We have tested it in 5.4.51-v7l+ kernel version.

oiluigio, 2020-10-20:
Hi, thanks for your reply.
My device has now a fresh installed os 5.4.51-v7l+ kernel version (I formatted my sd card and reinstall).
I reinstall MIPI_camera as well following the README, but the problem seems to be still there, everything freezes after any command...
[image]

any suggestions?

oiluigio, 2020-10-21:
I also run ./camera_i2c from utils folder as suggested in the troubleshooting and this is the output
[image]

oiluigio, 2020-11-03:
Here is an update, 
I connected my pi4 to a screen and realised that everything works fine, except that the ethernet connection disappear for some reason, which explain why everything was freezing when trying to use any of the routines via ssh.
My actual turnaround is using the wifi connection, but in the end I would like to use the ethernet connection again...
Here is a screenshot of ifconfig output before and after I run capture.c, as you can see the ip has disapperared from the eth0 connection.

 
[image]

gernot-hochmuth, 2020-11-21:
Hi, 
I encountered the same problem using a raspberry pi 4B. The wired ethernet connection gets broken until a reboot after accessing the MIPI-camera (either using the list_format utility or using arducamstill). This happens when using the latest wiringpi library (version 2.52). @stefanschuermans suggested a work-around in #49. The issue does not appear when using the wiringpi library provided by the raspbian repository (wiringpi 2.50). Using this version of wiringpi I can access my MIPI-camera and the ethernet connection stays alive. 
If you had installed "wiringpi-latest.deb" previously by 
```Shell
wget https://project-downloads.drogon.net/wiringpi-latest.deb
sudo dpkg -i wiringpi-latest.deb
```
 (as suggested by the installation instructions of the MIPI SDK library for the RPI 4B) you can try to uninstall it by
`sudo apt purge wiringpi`
and install wiringpi from the repository instead by
`sudo apt install wiringpi` 
afterwards. 
Check the wiringpi version by typing `gpio -v` in a terminal. The gpio utility provided by this version of wiringpi will not recognize the type of the raspberry pi model 4B properly. But it seems to work anyway without breaking the ethernet connection.
good luck!

oiluigio, 2020-11-21:
It did work! Ethernet connection stay up now.

Really appreciate!

closed 2020-11-21 by oiluigio