ArduCAM/MIPI_Camera #70: Cannot install IMX477 driver on Jetson Nano

lukaszungier, 2020-10-18:
Hi, I have a problem during installation of driver for IMX477 on Jetson Nano.
Here's what I got:
<pre><font color="#8AE234"><b>lukasz@lukasz-jetson</b></font>:<font color="#729FCF"><b>~/Downloads</b></font>$ sudo dpkg -i arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200904112010_arm64_imx477.deb 
Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 141866 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200904112010_arm64_imx477.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200904112010) ...
<b>dpkg:</b> dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213809); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.3-20200924161615.

<b>dpkg:</b> error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel
</pre>

Here's some information about my system:
<pre><font color="#8AE234"><b>lukasz@lukasz-jetson</b></font>:<font color="#729FCF"><b>~</b></font>$ cat /etc/nv_tegra_release 
# R32 (release), REVISION: 4.3, GCID: 21589087, BOARD: t210ref, EABI: aarch64, DATE: Fri Jun 26 04:38:25 UTC 2020
<font color="#8AE234"><b>lukasz@lukasz-jetson</b></font>:<font color="#729FCF"><b>~</b></font>$ uname -a
Linux lukasz-jetson 4.9.140-tegra #1 SMP PREEMPT Thu Sep 24 16:07:04 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux
</pre>

I've already tried _dpkg --confifure -a_ and _sudo apt-get install -f_
Could you help me?

glddiv, 2020-10-20:
Hi @lukaszungier 
Are you using SDCard Image to install the system? I tried to download the latest SDCard Image and did not encounter such problems, so I guess you installed the system using NVIDIA SDK Manager. If this is the case, you can refer to this question Solution: #68

GTRedE, 2020-10-22:
> Hi @lukaszungier
> Are you using SDCard Image to install the system? I tried to download the latest SDCard Image and did not encounter such problems, so I guess you installed the system using NVIDIA SDK Manager. If this is the case, you can refer to this question Solution: #68

I am running into the same problem as the OP on Nano. Before any updates I get an even newer revision: 4.4.

glddiv, 2020-10-26:
Hi @GTRedE 
l4t32.4.4 has been updated
https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.1

GTRedE, 2020-10-27:
> Hi @GTRedE 
> 
> l4t32.4.4 has been updated
> 
> https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.1

Thank you! But once again, since we cannot download prior versions of the Sd images, I think it might make make sense to include the working the images as part of the git. But just an opinion.

ArduCAM, 2020-10-27:
@GTRedE 
It is too large to keep every SD images on the github.
You can either backup the old SD images or use the latest one with the latest driver.

ddonco, 2020-10-27:
You can find previous Jetson Nano images on the nvidia site: [https://developer.nvidia.com/embedded/downloads](https://developer.nvidia.com/embedded/downloads)
Just do a quick ctrl+f for Jetson Nano and start scrolling down that page. The JP 4.4 2020/07/07 image seems to be compatible with the arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200904112010_arm64_imx477.deb driver. It worked for me.

ArduCAM, 2020-10-27:
@ddonco 
Thanks for your information.