ArduCAM/MIPI_Camera #73: Can not open video

galdalali, 2020-11-03:
Hey there,

I'm trying to open IMX477 mini HQ camera on my Jetson nano.
I installed the driver using [this tool](https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.2) and it seems to run OK
Tried 2 different Jetson's with 2 different cameras and I got the same issue when trying to display the video using gstreamer example:

```
gst-launch-1.0 nvarguscamerasrc sensor-id=$SENSOR_ID ! "video/x-raw(memory:NVMM),width=1920,height=1080,framerate=$FRAMERATE/1" ! nvvidconv ! nvoverlaysink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
(Argus) Error FileOperationFailed: Failed socket read: Connection reset by peer (in src/rpc/socket/common/SocketUtils.cpp, function readSocket(), line 79)
(Argus) Error FileOperationFailed:  (propagating from src/rpc/socket/client/SocketClientDispatch.cpp, function openSocketConnection(), line 215)
(Argus) Error FileOperationFailed: Cannot create camera provider (in src/rpc/socket/client/SocketClientDispatch.cpp, function createCameraProvider(), line 102)
Error generated. /dvs/git/dirty/git-master_linux/multimedia/nvgstreamer/gst-nvarguscamera/gstnvarguscamerasrc.cpp, execute:551 Failed to create CameraProvider
Got EOS from element "pipeline0".
Execution ended after 0:00:15.166958703
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

Using "QT V4L2 test Utility" i'm getting all the right parameters with a black screen. instead of video

I'm using fresh [Jetson nano sd-img](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write) (nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.4)

Please help me solve this

glddiv, 2020-11-04:
Hi @galdalali 
Very strange phenomenon, I cannot reproduce your situation. I tried to google this problem and found a similar situation. Can you try it?
https://forums.developer.nvidia.com/t/libargus-fails-on-re-entry-if-killed/82388

galdalali, 2020-11-04:
**I tried to restart the service couple of times, no help..
This error related to the fact that the process enter a timeout:**

nvargus-daemon.service - Argus daemon
   Loaded: loaded (/etc/systemd/system/nvargus-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-11-04 08:06:31 IST; 4min 17s ago
 Main PID: 16028 (nvargus-daemon)
    Tasks: 41 (limit: 4182)
   CGroup: /system.slice/nvargus-daemon.service
           └─16028 /usr/sbin/nvargus-daemon

Nov 04 08:06:52 nvidia-desktop nvargus-daemon[16028]: SCF: Error Timeout:  (propagating from src/services/capture/CaptureServiceEvent.cpp, function wait(), line 59)
Nov 04 08:06:52 nvidia-desktop nvargus-daemon[16028]: Error: Camera HwEvents wait, this may indicate a hardware timeout occured,abort current/incoming cc
Nov 04 08:07:33 nvidia-desktop nvargus-daemon[16028]: === gst-launch-1.0[16078]: Connection closed (7F85C631D0)=== gst-launch-1.0[16078]: WARNING: CameraProvider was not destroyed before client connection
Nov 04 08:07:33 nvidia-desktop nvargus-daemon[16028]: SCF: Error InvalidState:  (propagating from src/services/gl/EGLStreamProducer.cpp, function ~EGLStreamProducer(), line 50)
Nov 04 08:07:38 nvidia-desktop nvargus-daemon[16028]: === gst-launch-1.0[16078]: WARNING: Cleaning up 1 outstanding stream settings...=== gst-launch-1.0[16078]: WARNING: Cleaning up 1 outstanding sessions
Nov 04 08:07:38 nvidia-desktop nvargus-daemon[16028]: waitForIdleLocked remaining request 103
Nov 04 08:07:38 nvidia-desktop nvargus-daemon[16028]: waitForIdleLocked remaining request 102
Nov 04 08:07:38 nvidia-desktop nvargus-daemon[16028]: waitForIdleLocked remaining request 101
Nov 04 08:07:38 nvidia-desktop nvargus-daemon[16028]: SCF: Error Timeout: waitForIdle() timed out (in src/api/Session.cpp, function waitForIdleLocked(), line 920)
Nov 04 08:07:38 nvidia-desktop nvargus-daemon[16028]: (Argus) Error Timeout:  (propagating from src/api/CaptureSessionImpl.cpp, function destroy(), line 166)

**Here is the dmesg log:**

[59154.683741] packetv4 coming ,,,
[59156.474457] imx477 6-001a: Error updating control ranges -22
[59156.495955] packetv4 coming ,,,
[59156.554291] misc tegra_camera_ctrl: tegra_camera_update_isobw: Warning, Requested ISO BW 3281250 has been capped to VI's max BW 1500000
[59156.651456] imx477 6-001a: Error updating control ranges
[59156.656912] ------------[ cut here ]------------
[59156.661534] WARNING: CPU: 3 PID: 16330 at drivers/media/v4l2-core/videobuf2-core.c:1350 vb2_start_streaming+0xdc/0x158
[59156.661560] Modules linked in: fuse bnep zram option usb_wwan overlay spidev userspace_alert simcom_wwan(O) nvgpu bluedroid_pm ip_tables x_tables

[59156.661566] CPU: 3 PID: 16330 Comm: qv4l2 Tainted: G        W  O    4.9.140-tegra #2
[59156.661568] Hardware name: NVIDIA Jetson Nano Developer Kit (DT)
[59156.661570] task: ffffffc0ae6aaa00 task.stack: ffffffc02fdbc000
[59156.661573] PC is at vb2_start_streaming+0xdc/0x158
[59156.661576] LR is at vb2_start_streaming+0x6c/0x158
[59156.661579] pc : [<ffffff8008b36c64>] lr : [<ffffff8008b36bf4>] pstate: 60400045
[59156.661580] sp : ffffffc02fdbfb30
[59156.661585] x29: ffffffc02fdbfb30 x28: 0000000000000000 
[59156.661589] x27: ffffffc0f5015800 x26: 0000000000000001 
[59156.661592] x25: ffffff8009fbef90 x24: 0000000000000012 
[59156.661596] x23: 0000000000000000 x22: ffffffc02fdbfd10 
[59156.661599] x21: ffffffc0f96cb768 x20: ffffffc0f96cb5d0 
[59156.661602] x19: 00000000ffffffea x18: 0000000000000010 
[59156.661605] x17: 0000000000000000 x16: 0000000000000000 
[59156.661608] x15: ffffffffffffffff x14: ffffff808a139a02 
[59156.661610] x13: ffffff800a139a10 x12: 0000000000000006 
[59156.661613] x11: 0000000005f5e0ff x10: 00000000000038f8 
[59156.661616] x9 : 00000000ffffffd0 x8 : ffffff80083d3768 
[59156.661619] x7 : ffffff8009ea4198 x6 : ffffffc0feff7bf0 
[59156.661622] x5 : ffffffc0feff7bf0 x4 : 0000000000000000 
[59156.661624] x3 : ffffffc0feffd7f8 x2 : 0000000000000000 
[59156.661627] x1 : 0000000000000000 x0 : 0000000000000004 

[59156.661629] ---[ end trace 6e416b2342abf06e ]---
[59156.661631] Call trace:
[59156.661635] [<ffffff8008b36c64>] vb2_start_streaming+0xdc/0x158
[59156.661638] [<ffffff8008b389cc>] vb2_core_streamon+0x16c/0x1a0
[59156.661641] [<ffffff8008b3b58c>] vb2_streamon+0x54/0x88
[59156.661645] [<ffffff8008b3b614>] vb2_ioctl_streamon+0x54/0x60
[59156.661649] [<ffffff8008b157ec>] v4l_streamon+0x3c/0x50
[59156.661652] [<ffffff8008b1a94c>] __video_do_ioctl+0x204/0x2c8
[59156.661655] [<ffffff8008b1a2f8>] video_usercopy+0x2a0/0x6a0
[59156.661657] [<ffffff8008b1a734>] video_ioctl2+0x3c/0x50
[59156.661660] [<ffffff8008b14138>] v4l2_ioctl+0x88/0x118
[59156.661664] [<ffffff8008272158>] do_vfs_ioctl+0xb0/0x8d8
[59156.661667] [<ffffff8008272a0c>] SyS_ioctl+0x8c/0xa8
[59156.661670] [<ffffff8008083900>] el0_svc_naked+0x34/0x38
[59158.408221] packetv4 coming ,,,

glddiv, 2020-11-13:
Hi @galdalali
Sorry for the late reply, can you try to flash a new SD?

jung-youjin, 2021-11-07:
Hi @galdalali is this solved? Facing the same issue rn 😢

galdalali, 2021-11-10:
@jung-youjin Di you try to run the automatic script installation? don't quit remember how this is solved but the new one works well - https://www.arducam.com/docs/camera-for-jetson-nano/native-jetson-cameras-imx219-imx477/imx477-how-to-install-the-driver/