ArduCAM/MIPI_Camera #74: support for 4.9.140-32.4.4

Amin-Golden, 2020-11-03:
I'm trying to install arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200716095813_arm64.deb driver on my  Jetson nano. but aim getting this

`Selecting previously unselected package arducam-nvidia-l4t-kernel.
(Reading database ... 177694 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.3-20200716095813_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.3-20200716095813) ...
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.3-20200625213809); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.4-20201016124427.

dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel`

`uname -a
Linux eagle 4.9.140-tegra #1 SMP PREEMPT Fri Oct 16 12:32:46 PDT 2020 aarch64 aarch64 aarch64 GNU/Linux`

`cat /etc/nv_tegra_release
# R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t210ref, EABI: aarch64, DATE: Fri Oct 16 19:44:43 UTC 2020`

glddiv, 2020-11-04:
Hi @Amin-Golden ,

The latest version has been moved to releases, you can also try to use the automatic installation script.

https://github.com/ArduCAM/MIPI_Camera/releases

Amin-Golden, 2020-11-04:
Thanks a lot for your fast response. 
I installed the ar2030 driver from this link,  Arducam camera is working fine, after installing this driver but my Rasberry pi camera stopped working and jetson could not find it in /dev anymore please help! (My jetson has two CSI ports and I am using raspberry pi in the second CSI) also, I think it somehow conflicts with the PCI-E driver, after this installation I have a problem with the boot.

luvwinnie, 2021-03-11:
@Amin-Golden Hi, my IMX477 camera is not working, could you tell me how did you install which one in order to make it work?