ArduCAM/MIPI_Camera #75: support for 4.9.140-tegra-32.4.4-20201027211359

jwhite2a, 2020-11-11:
Running the `install.sh` script on my Jetson Nano produced the following error:
```
Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.140-tegra-32.4.4-20201027211359
Jetson type:  Nano (Developer Kit Version)
```
Additionally, running `sudo dpkg -i arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201026143033_arm64.deb` for the package found [here](https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.1) also failed (as expected with the wrong version number): 
```Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.4-20201027211359.```

Will `4.9.140-tegra-32.4.4-20201027211359` be supported? Is there a workaround?

glddiv, 2020-11-11:
Hi @jwhite2a 
Did you install the system using NVIDIA SDK Manager?

jwhite2a, 2020-11-11:
I did not use the NVIDIA SDK Manager, I used a fresh SD card image found [here](https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit#write).

glddiv, 2020-11-12:
Hi @jwhite2a 
Thank you for your information, I will test it and add new support.

glddiv, 2020-11-12:
Hi @jwhite2a 
The current situation is a bit strange. I downloaded the latest sd card image, and its nvidia-l4t-kernel version is still 4.9.140-tegra-32.4.4-20201016124427. Do you do any other operations after installing the image?
Anyway, I generated a deb package based on version 20201027211359, you can try it.
[arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201027211359_arm64.zip](https://github.com/ArduCAM/MIPI_Camera/files/5530094/arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201027211359_arm64.zip)

bittnt, 2020-11-13:
Thanks @glddiv could you help me with generating a new one for 4.9.140-tegra-32.4.4-20201027211332? I tried the one you uploaded on a jetson nx with the latest stuff, it shows the following error message:
```
(Reading database ... 138311 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel_4.9.140-32.4.4-20201027211359_arm64.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.140-32.4.4-20201027211359) over (4.9.140-32.4.3-20201009160531) ...
ls: cannot access 'extlinux.conf_bak': No such file or directory
ls: cannot access 'Image.bak': No such file or directory
dpkg: dependency problems prevent configuration of arducam-nvidia-l4t-kernel:
 arducam-nvidia-l4t-kernel depends on nvidia-l4t-kernel (= 4.9.140-tegra-32.4.4-20201027211359); however:
  Version of nvidia-l4t-kernel on system is 4.9.140-tegra-32.4.4-20201027211332.

dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel
```

johanso-au, 2020-11-13:
Having the same issue with 4.9.140-tegra-32.4.4-20201027211332

```
Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.140-tegra-32.4.4-20201027211332
Jetson type:  Xavier NX (Developer Kit Version)
```

Are there plans for automatic builds to coincide with the kernel updates?

jwhite2a, 2020-11-13:
@glddiv The deb package based on version 20201027211359 you linked above worked perfectly for me. thank you! :+1: I appreciate the fast response.

To answer your question, I did run a `sudo apt-get update` and `upgrade`, which (probably) updated my version of `nvidia-l4t-kernel` to the most recent one. 

I'll also leave the issue open for now, to allow discussion about boarder tegra kernel support.

bittnt, 2020-11-13:
Thanks @jwhite2a . I am using Jetson nx, and the latest version is 4.9.140-tegra-32.4.4-20201027211332.

I tried upgrade, dist-upgrade, no luck so far.

jwhite2a, 2020-11-13:
I only ran `sudo apt-get update` and `sudo apt-get upgrade`. 

I've verified my fresh image Nano image from Nvidia had `nvidia-l4t-kernel` version 4.9.140-tegra-32.4.3-20200625213809.
Running the commands upgraded the version to 4.9.140-tegra-32.4.4-20201027211359.

One possible work around is looking at the possible upgrade/downgrade versions by running `sudo apt-cache showpkg nvidia-l4t-kernel`. I found versions supported by the `install.sh` script when I did this. Maybe you can down/upgrade the package with `sudo apt-get install nvidia-l4t-kernel=<version>`? This could be dangerous, I don't know.

glddiv, 2020-11-16:
Hello everyone, thank you very much for your information, I already know the reason why the kernel version does not correspond, and I will update the automatic installation script to adapt to the new kernel version soon.

glddiv, 2020-11-16:
Hello everyone, I just updated the automatic installation script to add support for the new version of the kernel. You can download the install.sh script again and try to perform the installation.

bittnt, 2020-11-16:
Hi @glddiv Thanks a lot. where is the new automatic installation script? I did not see any commits or update from this repo. Maybe I miss something.

glddiv, 2020-11-16:
Hi @bittnt 
To ensure that the link has not changed, I did not create a new release, you just need to re-download Install.sh.

mainakae, 2020-11-16:
Hi @glddiv I've just tried the installation script, and I got the same result than @jwhite2a:

Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.140-tegra-32.4.4-20201027211359
Jetson type:  Nano (Developer Kit Version)

glddiv, 2020-11-17:
Hi @mainakae 
I'm very sorry, I made a mistake, I confused the version numbers of NX and Nano, please download *.sh again.

closed 2020-11-17 by jwhite2a

mainakae, 2020-11-17:
Installed correctly, but still get no video device after plugin my raspberry pi hq camera (IMX477) into my Jetson Nano.

Thanks anyway!

mainakae, 2020-11-17:
Ok, I've read that the IMX477 needs a resistor to be removed in order to be compatible with Jetson Nano's I2C. I'll try that later, most surely that is the issue.

Thanks again and sorry for the trouble 👍 !

lars-nagel, 2023-12-27:
Would it be possible to update the script to work with newer kernel versions, as well?

Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  5.15.122-tegra-36.2.0-20231130105725
Jetson type:  NVIDIA Jetson Orin Nano Developer Kit