ArduCAM/MIPI_Camera #77: Stereo Demo Image Rectification Zoomed

Jeremy26, 2020-11-18:
Hello,

I managed to arrive at step 5 of Stereo Vision Demo -
the calibration results seems correct: however, the rectified images are too zoomed.

The line of code number 86 in the file 4_calibration.py is:
rectified_pair = calibration.rectify((imgLeft, imgRight))

After checking, it looks like the one that messes up.

Do you think the calibration is wrong and I should try again? The results for finding the checkboard were really good.

Thanks

glddiv, 2020-11-19:
Hi @Jeremy26 
There should be no problem with the calibration api, he called the opencv api internally, and most of these problems are in calibration.

Jeremy26, 2020-11-19:
I saw the API call. Then maybe, I should try taking pictures again.
However, it was really correct when I checked. So the calibration was fine.

Jeremy26, 2020-11-19:
Here's the result I got

[image]