ArduCAM/MIPI_Camera #78: Xavier NX video streaming issue

Davincb, 2020-11-24:
In low light I get the following image quality improving to a bluish tint with some clear picture with adequate lighting. Is this a board issue with the NX? Works better on my RPI4. 
[image]

Davincb, 2020-11-24:
The output in better lighting is blurred 
[image]

All connections are correct and I have followed the instructions on here: auto installing for my Jetson kernel and running with gstreamer

mikprin, 2020-12-01:
Try to change your aperture (open it a little).