ArduCAM/MIPI_Camera #79: IMX477 drivers not working on Jetson Xavier NX when booted from ssd

mikprin, 2020-11-30:
We were using IMX477 camera with Xavier NX and when it is booted from SD card drivers working and the camera also works as intended.
However, when root booted on SSD and I install drivers there the same way, this time `dmesg | grep imx477` shows no logs indicating that apparently that drivers are not working. I don't know if other people claiming their camera drivers are not working, dealing with the same issue, but is it possible to check this? I think it will help me and others a lot. We can provide any extra information relating my hardware and software configuration and any logs that are required.

glddiv, 2021-01-05:
Hi @mikprin 

if you install or uninstall the driver after booting from the SSD, it will not take effect, because the driver is installed in the SSD, but the system is still loaded from the SD Card when booting(Jetson Xavier NX does not support booting directly from SSD), so if you install the driver in the SSD, you need to manually replace the following files in the SD card:
1. /boot/arducam/
2. /boot/extlinux/extlinux.conf

mikprin, 2021-01-06:
> Hi @mikprin
>
> if you install or uninstall the driver after booting from the SSD, it will not take effect, because the driver is installed in the SSD, but the system is still loaded from the SD Card [...]

Thank you very much! I will try this solution soon and update bug status! Solution seems logical, I just didn't know drivers location in boot folder.

luvwinnie, 2021-03-12:
@mikprin Hi, are you able to run the camera with root from SSD?

mikprin, 2021-03-13:
Hi there! Yes it was really helpful solution! I hope opened issue will help others who face the same problem.

-- Михаил СоловьяновЛаборатория нейровычислительных систем МФТИ+79779057970

luvwinnie, 2021-03-13:
@mikprin Thank you so much! could you provide some instructions that how to do this?

luvwinnie, 2021-03-15:
> @mikprin Thank you so much! could you provide some instructions that how to do this?

Thank you! I have been able to replace the file in my SD Card to make the camera to work.