ArduCAM/MIPI_Camera #85: driver doesnt work with raspberry 4 / new python library

hstarmans, 2021-01-09:
The MIPI_camera leads to system kernel crashes on the raspberry 4. This can be mitigated by moving back to raspberry 3B
I have also created an example of a new python library for [Arducam](https://github.com/hstarmans/Arducampython).
This could be an example for the author @ArduCAM , on how to create a python library.
If your reading this, improve your C driver. If it fails, it kills the whole linux kernel and i need to reboot.
Secondly, clean up the repo. It takes a lot of time to download as you reupload new binaries all the time and push this into git.
This is not the purpose of git.
I do like the project, however.

glddiv, 2021-01-09:
Hi @hstarmans 

> The MIPI_camera leads to system kernel crashes on the raspberry 4. This can be mitigated by moving back to raspberry 3B

Thank you for your feedback. What camera are you using? We will run the test.

> I have also created an example of a new python library for Arducam.
This could be an example for the author @ArduCAM , on how to create a python library.

Thanks for the code. Initially we also considered packaging methods such as cython or pybinding, but for some reason, we chose ctypes. Your code can be a good start. thanks again.

> Secondly, clean up the repo. It takes a lot of time to download as you reupload new binaries all the time and push this into git.
Git is intended for text files, not for binaries.

We have also noticed this problem. Due to historical problems, many links in videos and documents point to this  repository. If we clean up, we will encounter some resistance.

> I do like the project, however.

Glad to hear that you like it, please let us know if you have any comments or suggestions.

glddiv, 2021-01-11:
Hi @hstarmans 
By the way, can you provide relevant error information? We don't seem to notice this problem.

hstarmans, 2021-01-11:
On the raspberry pi3b it works. On the raspberry pi 4, I have however;
```pi@raspberrypi:~/MIPI_Camera/RPI $ ./list_format
Found sensor ov2311 at address 60
```
Hereafter the systems crashes. This is not the case for the Raspberry pi3B.
Output of i2c detect. Note, that I modified wiringpi as it gave an issue.
```
setting GPIO for board revsion: d03114
Raspberry Pi3B / Pi3B+ / 3A / 4B(1G/2G/4G)
Set state of 133 to 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- 6c -- -- --
70: 70 -- -- -- -- -- -- --
```

Network error: Connection timed out

───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────

Session stopped
    - Press <return> to exit tab
    - Press R to restart session
    - Press S to save terminal output to file
     ┌────────────────────────────────────────────────────────────────────┐
     │                        • MobaXterm 20.6 •                          │
     │            (SSH client, X-server and networking tools)             │
     │                                                                    │
     │ ➤ SSH session to pi@192.168.1.175                                  │
     │   • SSH compression : ✔                                            │
     │   • SSH-browser     : ✔                                            │
     │   • X11-forwarding  : ✔  (remote display is forwarded through SSH) │
     │   • DISPLAY         : ✔  (automatically set on remote server)      │
     │                                                                    │
     │ ➤ For more info, ctrl+click on help or visit our website           │
     └────────────────────────────────────────────────────────────────────┘

Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Jan 11 19:55:43 2021

SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.

pi@raspberrypi:~ $ cd Arducampython/
pi@raspberrypi:~/Arducampython $ ls
build  camera_installer.egg-info  CMakeLists.txt  dist  LICENSE  README.md  setup.py  src  test
pi@raspberrypi:~/Arducampython $ cd ..
pi@raspberrypi:~ $ ls
Arducampython  Documents  firestarter  hexastorm_design  MIPI_Camera  Pictures  raspiraw   tmp     WiringPi
Desktop        Downloads  hexastorm    litex             Music        Public    Templates  Videos  wiringpi-latest.deb
pi@raspberrypi:~ $ cd MIPI_Camera/
pi@raspberrypi:~/MIPI_Camera $ ls
Jetson  README.md  RPI
pi@raspberrypi:~/MIPI_Camera $ cd RPI/
pi@raspberrypi:~/MIPI_Camera/RPI $ ls
1.jpg                  capture             capture_raw.c       list_format              ov9281_external_trigger.c  preview_setMode.c     read_write_sensor_reg    video2stdout.c
2.jpg                  capture2opencv      cvui.h              list_format.c            preview.c                  python                read_write_sensor_reg.c  video4cameras
2MPGlobalShuterDemo    capture2opencv.cpp  enable_i2c_vc.sh    Makefile                 preview-camera0            qrcode_detection      stereo_depth_demo        video4cameras.c
[... 211 lines cut ...]
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
dtoverlay=spi0-1cs
dtoverlay=spi1-1cs,cs0_pin=7

dtparam=i2c_vc=on
# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
start_x=1
gpu_mem=300      
```

glddiv, 2021-01-13:
Hi @hstarmans 
We tested it, but did not reproduce your problem.
[image]

Regarding the modification of Wiringpi you mentioned, there are known issues with Wiringpi on PI4, but we only need to upgrade Wiringpi to the latest.
[image]

Can you tell me your system version? Not sure if this is related to the system version.

hstarmans, 2021-01-13:
I upgraded the repo to the latest commit, installed the new binary and now It is fixed
```
commit 7f75cf1e0f870ba9eb567d71c5e09702b8a9f395 (HEAD -> master, origin/master, origin/HEAD)
Author: UCTRONICS <avahe@uctronics.com>
Date:   Wed Jan 13 09:05:18 2021 +0800

    Fix Ethernet problems
```
I am using
```
gpio version: 2.60
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 04, Memory: 8192MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.4
  * This Raspberry Pi supports user-level GPIO access.
```
Anyhow, thanks!

closed 2021-01-13 by hstarmans