ArduCAM/MIPI_Camera #86: OV9281 on RPI4

promistrio, 2021-01-09:
Hello! I have a fresh Raspbian buster and trouble when I try to run preview or another examples
```

pi@raspberrypi:~/MIPI_Camera/RPI $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/MIPI_Camera/RPI $ cd utils/
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ ./camera_i2c 
setting GPIO for board revsion: b03111
Raspberry Pi3B / Pi3B+ / 3A / 4B(1G/2G/4G)
Set state of 133 to 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- 64 -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ cd ../
pi@raspberrypi:~/MIPI_Camera/RPI $ ./preview-camera0 
Open camera...
init camera status = 4100
pi@raspberrypi:~/MIPI_Camera/RPI $ ./arducamstill 
Open camera...
Found sensor ov9281 at address 60
init camera status = 4100
pi@raspberrypi:~/MIPI_Camera/RPI $ ./capture2opencv 
Open camera...
Found sensor ov9281 at address 60
init camera status = 4100

```

yavta works well

tennisparty, 2021-01-30:
same problem on model 3A+
`pi@raspberrypi:~/MIPI_Camera/RPI $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/MIPI_Camera/RPI $ cd utils/
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ ./camera_i2c 
setting GPIO for board revsion: 9020e0
Raspberry Pi3B / Pi3B+ / 3A / 4B(1G/2G/4G)
Set state of 133 to 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ cd ../
pi@raspberrypi:~/MIPI_Camera/RPI $ ./preview-camera0 
Open camera...
init camera status = 4100
pi@raspberrypi:~/MIPI_Camera/RPI $ ./arducamstill 
Open camera...
Hardware platform: 9020e0
Found sensor ov9281 at address 60
init camera status = 4100
pi@raspberrypi:~/MIPI_Camera/RPI $ ./capture2opencv
Open camera...
Hardware platform: 9020e0
Found sensor ov9281 at address 60
init camera status = 4100
`

Yavta was working on the VL42 driver but I cant get it to work with this version of the driver running - `./yavta --capture=1000 -n 3 --encode-to=file.h264 -f UYVY -m -T /dev/video0`

tennisparty, 2021-01-30:
I am experiencing the same:

pi@raspberrypi:~/MIPI_Camera/RPI $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/MIPI_Camera/RPI $ cd utils/
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ ./camera_i2c 
setting GPIO for board revsion: 9020e0
Raspberry Pi3B / Pi3B+ / 3A / 4B(1G/2G/4G)
Set state of 133 to 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
pi@raspberrypi:~/MIPI_Camera/RPI/utils $ cd ../
pi@raspberrypi:~/MIPI_Camera/RPI $ ./preview-camera0 
Open camera...
init camera status = 4100
pi@raspberrypi:~/MIPI_Camera/RPI $ ./arducamstill 
Open camera...
Hardware platform: 9020e0
Found sensor ov9281 at address 60
init camera status = 4100
pi@raspberrypi:~/MIPI_Camera/RPI $ ./capture2opencv
Open camera...
Hardware platform: 9020e0
Found sensor ov9281 at address 60
init camera status = 4100

Yavta works ok for me on the standard VL42 driver - [https://www.arducam.com/docs/pi/ov9281/tutorial/#15-first-use](url)  but not with the MIPI driver

tennisparty, 2021-01-30:
Same issue on completely fresh install. There are some errors in the make install command during the installation

`pi@raspberrypi:~ $ cd MIPI_Camera/RPI/
pi@raspberrypi:~/MIPI_Camera/RPI $ make clean && make 
rm -f *.o
rm -f video4cameras arducamstill capture video list_format capture_raw raw_callback yuv_callback read_write_sensor_reg ov9281_external_trigger 2MPGlobalShuterDemo preview-camera0 preview-dualcam capture-dualcam video2stdout capture2opencv opencvGui qrcode_detection
gcc  -I. -g -O0 -std=gnu11 -o video4cameras video4cameras.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o arducamstill arducamstill.c -larducam_mipicamera -lpthread 
arducamstill.c: In function ‘main’:
arducamstill.c:467:17: warning: initialization of ‘char’ from ‘void *’ makes integer from pointer without a cast [-Wint-conversion]
     char path = NULL;//"./lens_shading_table/imx230/2672x2004.h";
                 ^~~~
arducamstill.c:468:44: warning: passing argument 2 of ‘arducam_set_lens_table’ makes pointer from integer without a cast [-Wint-conversion]
     arducam_set_lens_table(camera_instance,path );
                                            ^~~~
In file included from arducamstill.c:1:
arducam_mipicamera.h:230:67: note: expected ‘char *’ but argument is of type ‘char’
 int arducam_set_lens_table(CAMERA_INSTANCE camera_instance, char* lens_table_path);
                                                             ~~~~~~^~~~~~~~~~~~~~~
arducamstill.c:502:50: warning: passing argument 3 of ‘pthread_create’ from incompatible pointer type [-Wincompatible-pointer-types]
   int ret = pthread_create(&processCmd_pt, NULL, prcessCmd,&processData);
                                                  ^~~~~~~~~
In file included from arducamstill.c:9:
/usr/include/pthread.h:236:15: note: expected ‘void * (*)(void *)’ but argument is of type ‘void (*)(PROCESS_STRUCT *)’ {aka ‘void (*)(struct <anonymous> *)’}
       void *(*__start_routine) (void *),
       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
gcc  -I. -g -O0 -std=gnu11 -o capture capture.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o video video.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o list_format list_format.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o capture_raw capture_raw.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o raw_callback raw_callback.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o yuv_callback yuv_callback.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o read_write_sensor_reg read_write_sensor_reg.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o ov9281_external_trigger ov9281_external_trigger.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o 2MPGlobalShuterDemo 2MPGlobalShuterDemo.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o preview-camera0 preview-camera0.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o preview-dualcam preview-dualcam.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o capture-dualcam capture-dualcam.c -larducam_mipicamera -lpthread 
gcc  -I. -g -O0 -std=gnu11 -o video2stdout video2stdout.c -larducam_mipicamera -lpthread 
g++ -I. -g -std=gnu++11 -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -o capture2opencv capture2opencv.cpp -larducam_mipicamera -lpthread 
g++ -I. -g -std=gnu++11 -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -o opencvGui opencvGui.cpp -larducam_mipicamera -lpthread 
g++ -I. -g -std=gnu++11 -I/usr/include/opencv -lopencv_shape -lopencv_stitching -lopencv_superres -lopencv_videostab -lopencv_aruco -lopencv_bgsegm -lopencv_bioinspired -lopencv_ccalib -lopencv_datasets -lopencv_dpm -lopencv_face -lopencv_freetype -lopencv_fuzzy -lopencv_hdf -lopencv_line_descriptor -lopencv_optflow -lopencv_video -lopencv_plot -lopencv_reg -lopencv_saliency -lopencv_stereo -lopencv_structured_light -lopencv_phase_unwrapping -lopencv_rgbd -lopencv_viz -lopencv_surface_matching -lopencv_text -lopencv_ximgproc -lopencv_calib3d -lopencv_features2d -lopencv_flann -lopencv_xobjdetect -lopencv_objdetect -lopencv_ml -lopencv_xphoto -lopencv_highgui -lopencv_videoio -lopencv_imgcodecs -lopencv_photo -lopencv_imgproc -lopencv_core -o qrcode_detection qrcode_detection.cpp -larducam_mipicamera -lpthread  -lzbar`

tennisparty, 2021-02-09:
Now fixed

promistrio, 2021-02-14:
> Now fixed

Compilation was successful but now the same problem
```
Open camera...
Hardware platform: b03111
Found sensor ov9281 at address 60
init camera status = 4100
```

Did you use  Arducam_OBISP_MIPI_Camera_Module?

promistrio, 2021-02-14:
@ArduCAM could you tell me what the reason might be in my case? I use one OV9281 camera

tennisparty, 2021-02-18:
> > Now fixed
> 
> @tennisparty, Compilation was successful but now the same problem
> 
> ```
> Open camera...
> Hardware platform: b03111
> Found sensor ov9281 at address 60
> init camera status = 4100
> ```
> 
> Did you use Arducam_OBISP_MIPI_Camera_Module?
> 
> @tennisparty Have you got the MIPI driver working?

I did, Arudcam sent me a new libarducam.so file to use before compiling. I just tried a fresh install of this github version and it doesnt work without it. It is working again when i use that file. Best to chase them direct for the file.

chobitsfan, 2023-06-27:
Hi @tennisparty Could you be able to share how you reach arducam for updated libarducam_mipicamera.so? Thank you very much