ArduCAM/MIPI_Camera #87: jetson nano  IMX477 gstreamer no output from camera

ghavart, 2021-01-10:
No output from Gstreamer pipeline

- Arducam IMX477 [camera](https://www.uctronics.com/camera-modules/camera-for-nvidia/arducam-high-quality-camera-for-jetson-nano-12-3mp-1-2-3-inch-imx477-hq-camera-module-with-6mm-cs-mount-lens.html)
- Jetson Nano (4.9.140-tegra)
- Connection: MIPI CSI-2
- Driver: https://github.com/ArduCAM/MIPI_Camera/releases/tag/v0.0.1
- Gstreamer: 
1. gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! "video/x-raw(memory:NVMM),width=1920,height=1080,framerate=30/1" ! nvvidconv ! nvoverlaysink
2. gst-launch-1.0 nvarguscamerasrc sensor_id=0 ! 'video/x-raw(memory:NVMM),width=1920, height=1080, framerate=30/1, format=NV12' !  nvvidconv ! nvegltransform ! nveglglessink -e

In all cases: no visible output while the pipeline seems to be playing:  
```
Setting pipeline to PAUSED ...

Using winsys: x11 
Pipeline is live and does not need PREROLL ...
Got context from element 'eglglessink0': gst.egl.EGLDisplay=context, display=(GstEGLDisplay)NULL;
Setting pipeline to PLAYING ...
New clock: GstSystemClock

```

Can you please help?

P.S. IMX219 [camera](https://www.amazon.com/Jetson-Nano-Camera-IMX219-160-8-Megapixels/dp/B07T43K7LC) is working OK on the same pipeline.

glddiv, 2021-01-16:
Hi @ghavart 
It looks like you are using ssh to execute commands, please execute this command before running the camera:
export DISPLAY=:0

ghavart, 2021-01-16:
No. I was running it with a monitor attached on HDMI.
Here is another output:
`art@nano:~$ dmesg | grep imx477
[    1.230265] imx477 6-001a: tegracam sensor driver:imx477_v2.0.6
[    1.676021] vi 54080000.vi: subdev imx477 6-001a bound
art@nano:~$ gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! "video/x-raw(memory:NVMM),width=1920,height=1080,framerate=30/1" ! nvvidconv ! nvoverlaysink
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 4032 x 3040 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 22.250000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 22.250000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 1 
   Output Stream W = 1920 H = 1080 
   seconds to Run    = 0 
   Frame Rate = 59.999999 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:01:10.021821572
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
^C
art@nano:~$ dmesg | grep imx477
[    1.230265] imx477 6-001a: tegracam sensor driver:imx477_v2.0.6
[    1.676021] vi 54080000.vi: subdev imx477 6-001a bound
[  279.387645] imx477 6-001a: Error updating control ranges -22
[  279.398663] imx477 6-001a: Error updating control ranges -22
[  290.145881] imx477 6-001a: Error updating control ranges`

glddiv, 2021-01-16:
It seems normal from this log, can you provide the corresponding dmesg log

ghavart, 2021-01-16:
I believe my hardware version is A02. 
Here is the journalctl output on a similar experiment (very long file)
[dmesg.txt](https://github.com/ArduCAM/MIPI_Camera/files/5824016/dmesg.txt)

ghavart, 2021-01-16:
Sorry. Disregard the kernel log. Here is the full dmesg output:
[dmesg.txt](https://github.com/ArduCAM/MIPI_Camera/files/5824019/dmesg.txt)

glddiv, 2021-01-16:
Hi @ghavart 
Can you provide a picture of the hardware connection?

ghavart, 2021-01-16:
[image]
[image]

ghavart, 2021-01-16:
# R32 (release), REVISION: 4.4, GCID: 23942405, BOARD: t210ref, EABI: aarch64, DATE: Fri Oct 16 19:44:43 UTC 2020

glddiv, 2021-01-16:
Hi @ghavart 

Thank you very much for your information, I am undergoing a diagnosis.

ghavart, 2021-01-16:
Thanks @glddiv
I'll check on it tomorrow.

glddiv, 2021-01-16:
Hi @ghavart 
I reproduced the problem. The problem is at 1920x1080 resolution, and 4032x3040 is no problem. You can try a large resolution. Next, I will try to repair the driver.

glddiv, 2021-01-16:
Hi @ghavart 
The driver has been updated, please use install_imx477.sh to install the driver again.

ghavart, 2021-01-16:
Hi @glddiv Thank you very much! It works now. After the driver re-installation both 1920x1080 and 4032x3040 work.
Do you mind sharing what was the problem?
The only issue I saw is the solution works for HDMI only. On a DVI-connected monitor still has the same problem. But it certainly not a blocker for me. Thanks again!

closed 2021-01-16 by ghavart

glddiv, 2021-01-18:
Hi @ghavart 

Finally, it was found that the problem came out on the device tree. We developed it on B01. When testing on A02, only 4032x3040 was tested, and the device tree was not corrected.

> The only issue I saw is the solution works for HDMI only. On a DVI-connected monitor still has the same problem. But it certainly not a blocker for me. Thanks again!

Jetson Nano does not seem to have a DVI interface?
And the video output interface theoretically does not affect the MIPI interface.

ghavart, 2021-01-18:
Sorry. I meant Display Port on Nano. Thanks again!