ArduCAM/MIPI_Camera #89: IMX447 Pixel Binning Gstreamer Pipeline Jetson nano

Radifuddinahmed, 2021-01-21:
I need the gstreamer pipeline to use the pixel binning of IMX 447 for higher frame rates in my jetson nano.

can someone please share the gstreamer pipeline?

N.B. i will use that use that image in opencv in a python program

Thanks in advanced
[image]

ArduCAM, 2021-01-25:
sorry, we don't have this configuration at the moment.

Radifuddinahmed, 2021-01-25:
Will it be available?

iftahnaf, 2021-04-13:
Any update?

I'm trying to record a video with the IMX477 Arducam on my 2GB nano, maximum fps on 1080p is 60, any way to make it faster?
If I'll record a raw file could it be above 60fps?

Thanks!