ArduCAM/MIPI_Camera #90: Add compatibility Jetson Nano

devilwacause, 2021-01-24:
Cannot find the corresponding deb package, please send following information to support@arducam.com

Kernel version: 4.9.201-tegra-32.5.0-20210115145440
Jetson type: Nano (Developer Kit Version)

Thanks,

bruno15, 2021-01-25:
I have the same problem.

Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.201-tegra-32.5.0-20210115145440
Jetson type:  Nano (Developer Kit Version)

glddiv, 2021-01-27:
Hey guys,
We have updated the latest driver, you can download and execute install_imx477.sh again.

devilwacause, 2021-01-27:
Thanks will update tonight on how it goes (1-27-2021)

jwpleow, 2021-01-28:
Hi, I'm confused where to find this updated install script?

devilwacause, 2021-01-30:
@glddiv - sorry late - worked fine for me

EveraertJan, 2021-02-25:
Hi, 

I feel like the following error might be related

```
Cannot find the corresponding deb package, please send the following information to support@arducam.com
Kernel version:  4.9.201-tegra-32.5.1-20210219084526
Jetson type:  Nano (Developer Kit Version)
```

If it would be faster to downgrade anything, please let me know!

Thanks in advance