ArduCAM/MIPI_Camera #105: IMX477 red coloured when IR Cut filter is on with jetson nano?

luvwinnie, 2021-03-15:
I have bought a IMX477 with auto IR Cut Filter, When i turn off the light the IR cut filter is on, however the image was some how red coloured? It is a new one i think should not be a problem, i can't get the same image in this [video](https://www.youtube.com/watch?v=REnoLoT5S2o) which is showed as white and black colored? Is this maybe because of the raspistill with **awb greyworld** ?

luvwinnie, 2021-04-01:
does anyone can help me with this?

jasonchenPJ, 2021-06-11:
Hi @luvwinnie 
Did you solve the issue?
I am also curious how to get the white and black colored image.

I get the "red" image when awb=auto
[image]

but i get the "color" image when awb=greyworld 
[image]

luvwinnie, 2021-06-11:
No I haven't solve yet.

sheneman, 2023-02-28:
Did this get resolved?   We are having the same issue with the IMX477 on the Jetson Nano.  That is, in IR-mode all images are pink.   Full-light pictures are fine, but IR-mode are pink.   This is happening on multiple IMX477 cameras.

aaronwmorris, 2023-04-12:
I had this same issue on Linux with libcamera.  There is a different tuning file specifically the NOIR IMX477 camera.  In libcamera, the option to add is `--tuning-file /usr/share/libcamera/ipa/raspberrypi/imx477_noir.json`.

Perhaps there is a similar option on the Jetson.

jsalsburg, 2025-05-07:
The red color is normal. In low light, if you turn on AWB, you lose detail in the low light image; turn off AWB. The NoIR of the camera in low light will allow IR illumination which is Monochromatic. The imx477 sensor is different from most video chips, which are RGGB. the imx477 is RGBW which allows for better low light-low noise detection. Point the camera at a clear night sky, you will notice the camera sees better than your naked eyes.

jsalsburg, 2025-05-08:
Added fisheye lens to teststand camera.
[image]