ArduCAM/MIPI_Camera #107: ISP config error when using IMX477 CSI camera within docker

luvwinnie, 2021-03-16:
It shows the following invalid isp config when i use the IMX477 within the docker.

```
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 77: Error: Invalid isp config attribute: "sharpness.v5.tab.Preview[3]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 78: Error: Invalid isp config attribute: "sharpness.v5.tab.Still[0]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 79: Error: Invalid isp config attribute: "sharpness.v5.tab.Still[1]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 80: Error: Invalid isp config attribute: "sharpness.v5.tab.Still[2]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 81: Error: Invalid isp config attribute: "sharpness.v5.tab.Still[3]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 82: Error: Invalid isp config attribute: "sharpness.v5.tab.Video[0]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 83: Error: Invalid isp config attribute: "sharpness.v5.tab.Video[1]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 84: Error: Invalid isp config attribute: "sharpness.v5.tab.Video[2]={7,8,9,10,11,12,13}"
Mar 16 12:09:21 jetson-nano nvargus-daemon[16446]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 85: Error: Invalid isp config attribute: "sharpness.v5.tab.Video[3]={7,8,9,10,11,12,13}"
```

Anyone know how to fix?

luvwinnie, 2021-03-16:
I tried with this [link](https://www.arducam.com/docs/camera-for-jetson-nano/fix-red-tint-with-isp-tuning/) to change the camera_overrides.isp however it still show the below errors with nvargus-daemon.

```
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 34: Error: Invalid isp config attribute: "ltm.enableLf=TRUE"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 38: Error: Invalid isp config attribute: "ltm.strength={0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5,0.5}"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 39: Error: Invalid isp config attribute element: "awb.nightmode.enable=TRUE"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 40: Error: Invalid isp config attribute element: "awb.nightmode.gains={2.5,1,1,3.2}"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 41: Error: Invalid isp config attribute: "em.initialState=0"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 42: Error: Invalid isp config attribute: "em.thresholdLow=2080"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 43: Error: Invalid isp config attribute: "em.thresholdHigh=3232"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 47: Error: Invalid isp config attribute: "em.preset[0].wbgains={1.5,1.0,1.0,1.5}"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: NvCameraIspConfigFileLoad: type = 0x75 Config file "camera_overrides.isp" Line 51: Error: Invalid isp config attribute: "em.preset[1].wbgains={1.5,1.0,1.0,1.5}"
 3月 16 17:30:39 alps-jetson-nx nvargus-daemon[6999]: CAM: serial no file already exists, skips storing againLSC: LSC surface is not based on full res!
```

aTEraGOU, 2022-05-23:
Is there any news regarding these "Invalid isp config attribute" errors?

gorghino, 2022-12-15:
UP