ArduCAM/MIPI_Camera #123: arducam_mipicamera.py line 269  pointer error

fuzzybear62, 2021-07-18:
In python, using capture.py the program abort on the statement:

        frame = camera.capture(encoding = 'jpeg')
        frame.as_array.tofile("{}x{}.jpg".format(fmt[0],fmt[1]))

with error:

    frame.as_array.tofile("/data/jpg/{}x{}.jpg".format(fmt[0],fmt[1]))
  File "/usr/src/app/lib/arducam_mipicamera.py", line 269, in as_array
    return np.ctypeslib.as_array(self.buffer_ptr[0].data, shape=(self.length,))
ValueError: NULL pointer access

glddiv, 2021-07-19:
Hi @fuzzybear62 
Can you provide more information?
1. the camera used(SKU)
2. which mode is used
3. the Raspberry Pi model
4. a script that can reproduce the problem.

fuzzybear62, 2021-07-19:
The camera is the imx298 and the script is capture.py in your examples.
RPI 3+

glddiv, 2021-07-20:
Hi @fuzzybear62
Have you modified the resolution or mode in capture.py?
Does the null pointer problem often occur?
At present, I have tried several times and have not reproduced the problem. Maybe some special conditions are needed?

fuzzybear62, 2021-07-20:
Hi,

no, used capture.py as is

null pointer every time, no way to make it working. The only way is to 
use a older version of libarducam.so

Yes, probably some special condition that is not easy to identify are needed

ArduCAM, 2021-07-20:
Have you tested it using arducamstill tool with the latest libarducam_mipicamera.so ?
What's more, Please tell me which version lib you are using.

minewilliam, 2021-10-13:
I downloaded the latest version (Sept 25 2021 on commit: 11de396bfbd3cfc41ab20bf92a4bbbcb08294210)
I too am using the imx298 sensor (bought 2 years ago in 2019) on a raspberry pi 4B
I ran arducamstill like shown below. It took roughly a minute or two (not 5 seconds) for the script to exit with `capture timeout`.
The test.jpg file was not created.
```
pi@raspberry:~/MIPI_Camera/RPI $ ./arducamstill -t 5000 -m 1 -e jpg -o test.jpg
Open camera...
Found sensor imx298 at address 1A
mode: 0, width: 1160, height: 800, pixelformat: pBAA, desc: 4x4 binning 60fps
mode: 1, width: 1920, height: 1080, pixelformat: pBAA, desc: 60fps
mode: 2, width: 2336, height: 1748, pixelformat: pBAA, desc: 2x2 binning 27fps
mode: 3, width: 4656, height: 3496, pixelformat: pBAA, desc: 8fps
mode: 4, width: 4656, height: 3496, pixelformat: pBAA, desc: 12fps
index: 0, CID: 0x00980914, desc: V4L2_CID_HFLIP, min: 0, max: 1, default: 0, current: 0
index: 1, CID: 0x00980915, desc: V4L2_CID_VFLIP, min: 0, max: 1, default: 0, current: 0
index: 2, CID: 0x00980911, desc: V4L2_CID_EXPOSURE, min: 0, max: 65535, default: 1660, current: 1000
index: 3, CID: 0x009A090A, desc: V4L2_CID_FOCUS_ABSOLUTE, min: 0, max: 1023, default: 150, current: 0
Start preview...
Current mode: 1, width: 1920, height: 1080, pixelformat: pBAA, desc: 60fps
mmal: Enable JPEG encoder.

capture timeout
```

As for the original issue, here is the output of capture.py:
```
pi@cyclop-pi:~/MIPI_Camera/RPI/python $ python3.7 capture.py 
Open camera...
Found sensor imx298 at address 1A
Setting the resolution...
mmal: Failed to fix lens shading, use the default mode!
Current resolution is (1920, 1080)
Start preview...
Reset the focus...
Enable Auto Exposure...
Enable Auto White Balance...
mmal: Enable JPEG encoder.

NULL pointer access
```
I tried with python v3.7.3 and v2.7.16