ArduCAM/MIPI_Camera #126: OV9281 on RPi 4B cannot capture

rosterloh, 2021-08-16:
With a OV9281 (UC-599 Rev.B) board connected to a Raspberry Pi 4B all of these applications hang while trying to capture image data.
```bash
$ uname -a
Linux raspberrypi 5.10.52-v7l+ #1441 SMP Tue Aug 3 18:11:56 BST 2021 armv7l GNU/Linux
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
$ ./capture
Open camera...
Found sensor ov9281 at address 60
mode: 0, width: 1280, height: 800, pixelformat: GREY, desc:  1 lan raw8 60fps
mode: 1, width: 1280, height: 720, pixelformat: GREY, desc: (null)
mode: 2, width: 640, height: 400, pixelformat: GREY, desc: (null)
mode: 3, width: 320, height: 200, pixelformat: GREY, desc: (null)
mode: 4, width: 160, height: 100, pixelformat: GREY, desc: (null)
mode: 5, width: 1280, height: 800, pixelformat: GREY, desc: Used for ov9281 2lans raw8
mode: 6, width: 1280, height: 800, pixelformat: Y10P, desc: Used for ov9281 2lans raw10
mode: 7, width: 1280, height: 800, pixelformat: GREY, desc: Used for ov9281 1lane raw8 1280x800 external trigger mode
mode: 8, width: 1280, height: 720, pixelformat: GREY, desc: Used for ov9281 1lane raw8 1280x720 external trigger mode
mode: 9, width: 640, height: 400, pixelformat: GREY, desc: Used for ov9281 1lane raw8 640x400 external trigger mode
mode: 10, width: 320, height: 200, pixelformat: GREY, desc: Used for ov9281 1lane raw8 320x200 external trigger mode
mode: 11, width: 1280, height: 800, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 1280x800 external trigger mode
mode: 12, width: 1280, height: 800, pixelformat: Y10P, desc: Used for ov9281 2lanes raw10 1280x800 external trigger mode
mode: 13, width: 1280, height: 720, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 1280x720 external trigger mode
mode: 14, width: 640, height: 400, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 640x400 external trigger mode
mode: 15, width: 320, height: 200, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 320x200 external trigger mode
mode: 16, width: 1280, height: 800, pixelformat: BA81, desc:  Used for ov9782 1 lan raw8 60fps
mode: 17, width: 1280, height: 720, pixelformat: BA81, desc: Used for ov9782 1280x720
mode: 18, width: 640, height: 400, pixelformat: BA81, desc: Used for ov9782 640x400
mode: 19, width: 320, height: 200, pixelformat: BA81, desc: Used for ov9782 320x200
mode: 20, width: 160, height: 100, pixelformat: BA81, desc: Used for ov9782 160x100
mode: 21, width: 1280, height: 800, pixelformat: BA81, desc: Used for ov9281 2lans raw8
mode: 22, width: 1280, height: 800, pixelformat: pBAA, desc: Used for ov9782 1280x800
index: 0, CID: 0x00980914, desc: V4L2_CID_HFLIP, min: 0, max: 1, default: 0, current: 0
index: 1, CID: 0x00980915, desc: V4L2_CID_VFLIP, min: 0, max: 1, default: 0, current: 0
index: 2, CID: 0x00980911, desc: V4L2_CID_EXPOSURE, min: 1, max: 65535, default: 681, current: 32
index: 3, CID: 0x00980913, desc: V4L2_CID_GAIN, min: 0, max: 15, default: 1, current: 8
index: 4, CID: 0x00981901, desc: V4L2_CID_ARDUCAM_EXT_TRI, min: 0, max: 1, default: 0, current: 0
Please choose sensor mode: 0
Setting the mode...
choose the mode 0

```
No further connection can be made to the sensor after this point until the device is rebooted

rosterloh, 2021-08-24:
@ArduCAM I'd really appreciate a response to this. We have an active piece of development on going with yourselves (UC-865) and I can't continue my validation if this library doesn't work. Is there a debug version of the library I could use to give you more information or any other steps you'd like me to perform?

rosterloh, 2021-08-24:
I'm sure this is not very helpful but here is the output run with gdb
```bash
pi@raspberrypi:~/MIPI_Camera/RPI $ gdb -ex=r --args arducamstill -t 0 -m 0 -e png -o test.png
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from arducamstill...done.
Starting program: /home/pi/MIPI_Camera/RPI/arducamstill -t 0 -m 0 -e png -o test.png
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Open camera...
[New Thread 0xb6c5a460 (LWP 831)]
[New Thread 0xb62ff460 (LWP 832)]
[New Thread 0xb5afe460 (LWP 833)]
[New Thread 0xb52fd460 (LWP 834)]
[Detaching after fork from child process 835]
[Detaching after fork from child process 839]
[Detaching after fork from child process 841]
[Detaching after fork from child process 843]
[Detaching after fork from child process 845]
[Detaching after fork from child process 847]
Found sensor ov9281 at address 60
[New Thread 0xb4afc460 (LWP 849)]
[New Thread 0xb42fb460 (LWP 850)]
[New Thread 0xb3afa460 (LWP 851)]
[New Thread 0xb32f9460 (LWP 852)]
[New Thread 0xb2af8460 (LWP 853)]
mode: 0, width: 1280, height: 800, pixelformat: GREY, desc:  1 lan raw8 60fps
mode: 1, width: 1280, height: 720, pixelformat: GREY, desc: (null)
mode: 2, width: 640, height: 400, pixelformat: GREY, desc: (null)
mode: 3, width: 320, height: 200, pixelformat: GREY, desc: (null)
mode: 4, width: 160, height: 100, pixelformat: GREY, desc: (null)
mode: 5, width: 1280, height: 800, pixelformat: GREY, desc: Used for ov9281 2lans raw8
mode: 6, width: 1280, height: 800, pixelformat: Y10P, desc: Used for ov9281 2lans raw10
mode: 7, width: 1280, height: 800, pixelformat: GREY, desc: Used for ov9281 1lane raw8 1280x800 external trigger mode
mode: 8, width: 1280, height: 720, pixelformat: GREY, desc: Used for ov9281 1lane raw8 1280x720 external trigger mode
mode: 9, width: 640, height: 400, pixelformat: GREY, desc: Used for ov9281 1lane raw8 640x400 external trigger mode
mode: 10, width: 320, height: 200, pixelformat: GREY, desc: Used for ov9281 1lane raw8 320x200 external trigger mode
mode: 11, width: 1280, height: 800, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 1280x800 external trigger mode
mode: 12, width: 1280, height: 800, pixelformat: Y10P, desc: Used for ov9281 2lanes raw10 1280x800 external trigger mode
mode: 13, width: 1280, height: 720, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 1280x720 external trigger mode
mode: 14, width: 640, height: 400, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 640x400 external trigger mode
mode: 15, width: 320, height: 200, pixelformat: GREY, desc: Used for ov9281 2lanes raw8 320x200 external trigger mode
mode: 16, width: 1280, height: 800, pixelformat: BA81, desc:  Used for ov9782 1 lan raw8 60fps
mode: 17, width: 1280, height: 720, pixelformat: BA81, desc: Used for ov9782 1280x720
mode: 18, width: 640, height: 400, pixelformat: BA81, desc: Used for ov9782 640x400
mode: 19, width: 320, height: 200, pixelformat: BA81, desc: Used for ov9782 320x200
mode: 20, width: 160, height: 100, pixelformat: BA81, desc: Used for ov9782 160x100
mode: 21, width: 1280, height: 800, pixelformat: BA81, desc: Used for ov9281 2lans raw8
mode: 22, width: 1280, height: 800, pixelformat: pBAA, desc: Used for ov9782 1280x800
index: 0, CID: 0x00980914, desc: V4L2_CID_HFLIP, min: 0, max: 1, default: 0, current: 0
index: 1, CID: 0x00980915, desc: V4L2_CID_VFLIP, min: 0, max: 1, default: 0, current: 0
index: 2, CID: 0x00980911, desc: V4L2_CID_EXPOSURE, min: 1, max: 65535, default: 681, current: 32
index: 3, CID: 0x00980913, desc: V4L2_CID_GAIN, min: 0, max: 15, default: 1, current: 8
index: 4, CID: 0x00981901, desc: V4L2_CID_ARDUCAM_EXT_TRI, min: 0, max: 1, default: 0, current: 0
^C
Thread 1 "arducamstill" received signal SIGINT, Interrupt.
0xb6f2e088 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=1, futex_word=0xb6cf73c4 <client+36>) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
205	../sysdeps/unix/sysv/linux/futex-internal.h: No such file or directory.
(gdb) 
```

ArduCAM, 2021-09-07:
Dear Sir,

This is Bin form Arducam support team. I am terrible sorry for replying you so late. Don't worry and  I will try my best to help you.
In order  to solve your issue in time. Please send  your issue to our support email the email id is support@arducam.com.
Recently, we have updated our lib, please download our latest lib to test and tell me your latest result.
Feel free to let me know if you need more help.
Bin from Arducam support team.