ArduCAM/MIPI_Camera #143: IMX477 Camera not detected on Jetson Xavier NX

weeshal, 2022-02-03:
Hi, we have 2 IMX477 cameras connected through CSI, and recently have not been able to detect and view feeds of 1 of them. We switched wires and cameras, but the camera is not able to be detected through the specific CSI connected on the Jetson - was hoping to get further troubleshooting steps to confirm if the issue is with the connector or camera. I am attaching a few steps we have already attempted
Here is a screenshot of dmesg logs:
[image]

and the error message generated through gst-launch-1.0:
[image]

Please let me know what further steps I can take to identify the root cause of these errors.

thumbcore, 2022-07-27:
You might consider putting your console outputs in the GitHub issue as text, instead of an image. This allows others to search for your log output with Google. and can make it easier to read.

See [GitHub docs](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks)

For example:

```console
$ dmesg | grep imx
imx477 9-001a: tegracam sensor driver:imx477_v2.0.6
imx477 10-001a: tegracam sensor driver:imx477_v2.0.6
imx477 10.001a: imx477_board_setup: error during i2c read probe (-5)
imx477 10-001a: board setup failed
imx477: probe of 10-001a failed with error -5
tegra194-vi5 15c10000.vi: subdev imx477 9-001a bound
```

tugbakara, 2023-03-30:
I have the same issue in Jetson Nano with imx519, it doesn't seem to be detected