ArduCAM/MIPI_Camera #163: Xavier NX camera wont work

bajpaia, 2022-11-04:
Imx477 camera
r35 Xavier NX module

used the automatic installation script 
logs for dmesg |grep imx477
[   11.046104] nv_imx477: no symbol version for module_layout
[   11.046877] nv_imx477: loading out-of-tree module taints kernel.
[   11.048512] imx477 9-001a: tegracam sensor driver:imx477_v2.0.6
[   11.358117] imx477 9-001a: imx477_board_setup: error during i2c read probe (-121)
[   11.359303] imx477 9-001a: board setup failed
[   11.360387] imx477: probe of 9-001a failed with error -121
[   11.362041] imx477 10-001a: tegracam sensor driver:imx477_v2.0.6
[   11.663739] tegra-camrtc-capture-vi tegra-capture-vi: subdev imx477 10-001a bound

ls /dev/video* gives output /dev/video0

hunter55555, 2024-08-12:
I had the same problem. Did you solve it? How? thanks in advance.

hunter55555, 2024-08-13:
After analysis, `imx477 9-001a: imx477_board_setup: error during i2c read probe (-121) imx477 9-001a: board setup failed` is not an error, this is because cam0 is connected. If cam1 is connected, the message  `imx477 10-001a: imx477_board_setup: error during i2c read probe (-121) imx477 10-001a: board setup failed` will appear. You can refer to [this website](https://developer.ridgerun.com/wiki/index.php?title=Raspberry_Pi_HQ_camera_IMX477_Linux_driver_for_Jetson#Example_Pipelines) to run Example Pipelines.