ArduCAM/MIPI_Camera #169: support for imx519 with jetson nano 2GB?

fpgasdr, 2023-02-09:
hi in documentation you says for install this camera:

https://docs.arducam.com/Nvidia-Jetson-Camera/Native-Camera/Quick-Start-Guide/

Step 1. Download the bash scripts

cd ~
wget https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/install_full.sh

Step 2. Install the driver

chmod +x install_full.sh
./install_full.sh -m imx519

but with mi jetson nano I get:

Unsupported modules.

Supported modules:
	imx477_stereo
	arducam
	imx477_master_slave
	imx477
	imx230
	imx298

how should be the installation steps?.

tugbakara, 2023-03-30:
I have installed IMX519 driver into my 2GB Nano but the camera cannot be detected. Did you solve your issue?

skymeson, 2023-07-03:
Hi, I''m also having an issue getting the imx519 working with Nano 2Gb. I have installed the drivers and followed the Quick Start guide, but when I try running the displayer.py script, I get an error. 

python3 arducam_displayer.py -f Y16 -d 0 --fps
Hardware is: NVIDIA Jetson Nano (2 GB ram)
[ WARN:0@12.910] global cap_v4l.cpp:1119 tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.
Traceback (most recent call last):
  File "arducam_displayer.py", line 152, in <module>
    display(cap, arducam_utils, args.fps)
  File "arducam_displayer.py", line 66, in display
    frame = resize(frame, 1280.0)
  File "arducam_displayer.py", line 44, in resize
    width = frame.shape[1]
AttributeError: 'NoneType' object has no attribute 'shape'

The output of v4l2-ctl is 
v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'RG10'
	Name        : 10-bit Bayer RGRG/GBGB
		Size: Discrete 4656x3496
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 3840x2160
			Interval: Discrete 0.048s (21.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.008s (120.000 fps)

and my dmesg output is 

dmesg | grep imx
[    0.212340] DTS File Name: /var/jenkins_home/workspace/rnel_l4t-32.7.1-imx519-low-speed/kernel/kernel-4.9/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t210/batuu/kernel-dts/tegra210-p3448-0003-p3542-0000.dts
[    0.412385] DTS File Name: /var/jenkins_home/workspace/rnel_l4t-32.7.1-imx519-low-speed/kernel/kernel-4.9/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t210/batuu/kernel-dts/tegra210-p3448-0003-p3542-0000.dts
[    1.275768] imx519 6-001a: tegracam sensor driver:imx519_v2.0.6
[    1.733558] vi 54080000.vi: subdev imx519 6-001a bound