ArduCAM/MIPI_Camera #180: IMX477 driver is not installing in Jetson Orin

Eldho1416, 2023-07-19:
Hi Everyone,

I have camera [https://robu.in/product/arducam-high-quality-ir-cut-camera-for-jetson-nano-xavier-nx-12-3mp-1-2-3-inch-imx477-hq-camera-module-with-6mm-cs-lens/](url) and i tried installing this driver but i'm not able to to install.

Error: 
```
(Reading database ... 155210 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel-t234-nx-5.10.104-tegra-35.3.1-20230707043543_arm64_imx477.deb ...
Unpacking arducam-nvidia-l4t-kernel (5.10.104-tegra-35.3.1-20230707043543) over (5.10.104-tegra-35.3.1-20230707043543) ...
Setting up arducam-nvidia-l4t-kernel (5.10.104-tegra-35.3.1-20230707043543) ...

Traceback (most recent call last):
File \"/opt/arducam/jetson-io/config-by-hardware.py\", line 125, in
main()
File \"/opt/arducam/jetson-io/config-by-hardware.py\", line 94, in main
jetson = board.Board()
File \"/opt/arducam/jetson-io/Jetson/board.py\", line 230, in __init__
self.dtb = _board_get_dtb(self.compat, self.model, dtbdir)
File \"/opt/arducam/jetson-io/Jetson/board.py\", line 114, in _board_get_dtb
raise RuntimeError(\"No DTB found for %s!\" % model)
RuntimeError: No DTB found for NVIDIA Orin NX Developer Kit!
An unknown error occurred while installing dtoverlays.
dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
installed arducam-nvidia-l4t-kernel package post-installation script subprocess returned error exit status 255
Errors were encountered while processing:
arducam-nvidia-l4t-kernel
```

Dion4cen, 2023-08-04:
Hi，
Looks like you are using the third-party carrier board.

gustavosousa2208, 2023-08-09:
I had the same error, but the problem is with the kernel version. I have Orin Nano with kernel 5.10.120 and the release is in version 5.10.104, if installing the lesser versions, the packages dont match version and dont install. If i get the major version, but for other Jetson it gives me the same error. Is there a way of installing 104 driver for 120 with most updated packages?

Euncheol-Kang, 2023-09-15:
@gustavosousa2208 Have you solved the problem? I think I have the same problem.... OTL

Eldho1416, 2023-12-23:
Hey, sorry for the late reply. Even i couldn't solve this issue. 

If you're only looking to enable the driver you can follow this documentation. https://wiki.seeedstudio.com/reComputer_Industrial_J40_J30_Hardware_Interfaces_Usage/#usage