ArduCAM/MIPI_Camera #186: Cannot install Ar0234 driver in Jetson Orin nano (L4T 35.4.1 Jetpack 5.1.2)

jackwei86, 2023-11-30:
hi.there
  I have checked the dtb file genrated by driver(arducam-nvidia-l4t-kernel-t234-nano-5.10.120-tegra-35.4.1-20230809074912_arm64.deb). The dtb file is not match my Orin nano .

The prefix name of driver generated dtb file   is  tegra234-p3767-0003-p4768-0000-a0-user-custom.dtb
My current board used is tegra234-p3767-0004-p4768-0000-a0-user-custom.dtb.

It seems the driver is older than my current dtb version.

I have tried to modify the install python script (/opt/arducm/jeston-io/Utils/dtc.py  line 117 ) to  skip dtb compatible check and install the driver successfully.After reboot the board ,i run the script /opt/jetson-io/jetson-io.py to configure the  24pin  csi connector and save changes. 
After all this steps,the cmd "sudo v4l2-ctl --list-devices" show "cannot open device /dev/video0" , it seems  the board still can not find the camera connected.

Should i wait for any version updates or if anyone has suggestions?

Thanks for any replies.

jackwei86, 2023-12-05:
I have tried to install the  `arducam-nvidia-l4t-kernel-t234-nano-5.10.120-tegra-35.4.1-20230809074912_arm64.deb` in my new Jetson Orin 8G which the P-Number is p3767-0003. The carrier dtb version is exactly matching the driver required. But the camera cannot work. The cmd  "sudo v4l2-ctl --list-devices"  still shows "cannot open device /dev/video0,exiting".

Dion4cen, 2024-01-11:
> The cmd "sudo v4l2-ctl --list-devices" still shows "cannot open device /dev/video0,exiting".
Can you connect this camrea to another camera port on your Jerson Orin carrier board?

Dion4cen, 2024-01-11:
Can you connect this camrea to another camera port on your Jerson Orin carrier board?

> The cmd "sudo v4l2-ctl --list-devices" still shows "cannot open device /dev/video0,exiting".

Dion4cen, 2024-01-11:
> I have tried to install the 
111

>