ArduCAM/MIPI_Camera #187: Need Orin nano support for certain Kernel version

linwk20, 2024-01-02:
Hi, 

I am using my Orin Nano with the stereo pack of OV9281:
https://www.arducam.com/product/arducam-1mp2-stereoscopic-camera-bundle-kit-for-raspberry-pi-nvidia-jetson-nano-xavier-nx-two-ov9281-global-shutter-monochrome-camera-modules-and-camarray-stereo-camera-hat/

But when I follow the tutorial (https://docs.arducam.com/Nvidia-Jetson-Camera/Multi-Camera-CamArray/quick-start/#for-arducam-jetvariety-camera) to install the driver running 
```
wget https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/install_full.sh
chmod +x install_full.sh
./install_full.sh -m arducam
```
It shows an error:

Cannot find the corresponding deb package, please send the following information to [support@arducam.com](mailto:support@arducam.com)
Kernel version:  5.15.122-tegra-36.2.0-20231130105725
Jetson type:  NVIDIA Jetson Orin Nano Developer Kit

Here are my questions:

     (1) Can you provide a jetvariety of drivers for above version(5.15.122-tegra-36.2.0-20231130105725)? If yes, how long would it take, I am catching a deadline so if it takes long I will downgrade my kernel.

     (2) Or is it possible that I can build it myself? Can you tell me how to do it? I did not see related things in the github repo

linwk20, 2024-01-02:
Got response from support team

closed 2024-01-02 by linwk20

hodoong3, 2025-02-08:
5.15.122-tegra-36.2.0-20231130105725
How did you get the drivers and is your camera working now?

linwk20, 2025-02-08:
they only support certain kernel, i have to downgrade OS of my orin nano.