ArduCAM/MIPI_Camera #188: Unknown error upon installing the driver for jetson nano imx708

Thesis123123, 2024-01-03:
I was trying to install the driver as per instructions in the [quick start guide](https://docs.arducam.com/Nvidia-Jetson-Camera/Native-Camera/Quick-Start-Guide/) but i got an error message on my terminal saying 

```
An unknown error occurred while installing dtoverlays.
dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 installed arducam-nvidia-l4t-kernel package post-installation script subprocess returned error exit status 255
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel

Unknown error, please send the error message to support@arducam.com
```

This is a freshly flashed OS nothing else installed or modified, jetpack version 4.9.3

Here is the full terminal from start to finish

```
jetson@jetson-desktop:~$ wget https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/install_full.sh
--2024-01-04 05:35:31--  https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/install_full.sh
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/ebdc0c54-a520-42e6-b4ef-e241b3acfd59?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240103T213531Z&X-Amz-Expires=300&X-Amz-Signature=109ac11d11e4809b9309a1934fea8515790d66c802733d8ce6c1520e0a8c325f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dinstall_full.sh&response-content-type=application%2Foctet-stream [following]
--2024-01-04 05:35:31--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/ebdc0c54-a520-42e6-b4ef-e241b3acfd59?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240103T213531Z&X-Amz-Expires=300&X-Amz-Signature=109ac11d11e4809b9309a1934fea8515790d66c802733d8ce6c1520e0a8c325f&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dinstall_full.sh&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4560 (4.5K) [application/octet-stream]
Saving to: ‘install_full.sh’

install_full.sh               100%[=================================================>]   4.45K  --.-KB/s    in 0.003s

2024-01-04 05:35:32 (1.44 MB/s) - ‘install_full.sh’ saved [4560/4560]

jetson@jetson-desktop:~$ chmod +x install_full.sh
jetson@jetson-desktop:~$ ./install_full.sh -m imx708
./install_full.sh: line 19: warning: command substitution: ignored null byte in input
NVIDIA Jetson Nano Developer Kit
--2024-01-04 05:35:49--  https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/modules.txt
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/8b494088-bb06-4531-875d-8d5d4ad9e7ae?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240103T213549Z&X-Amz-Expires=300&X-Amz-Signature=bdbf63b0c748e0df6778fd99f0ccb06990694cec3de2f78a0a5d64d44dd6649e&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dmodules.txt&response-content-type=application%2Foctet-stream [following]
--2024-01-04 05:35:49--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/8b494088-bb06-4531-875d-8d5d4ad9e7ae?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240103T213549Z&X-Amz-Expires=300&X-Amz-Signature=bdbf63b0c748e0df6778fd99f0ccb06990694cec3de2f78a0a5d64d44dd6649e&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dmodules.txt&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1346 (1.3K) [application/octet-stream]
Saving to: ‘modules.txt’

modules.txt                   100%[=================================================>]   1.31K  --.-KB/s    in 0.001s

2024-01-04 05:35:50 (1.27 MB/s) - ‘modules.txt’ saved [1346/1346]

--2024-01-04 05:35:50--  https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/imx708_links.txt
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/740a233c-3e22-4b32-8df4-7ce359a6e2c5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240103T213551Z&X-Amz-Expires=300&X-Amz-Signature=c833b9a3beb3145d939fe2b9299a2d09e29faa56c35e3c0bda6e5a1487fcd467&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dimx708_links.txt&response-content-type=application%2Foctet-stream [following]
--2024-01-04 05:35:51--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/740a233c-3e22-4b32-8df4-7ce359a6e2c5?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240103T213551Z&X-Amz-Expires=300&X-Amz-Signature=c833b9a3beb3145d939fe2b9299a2d09e29faa56c35e3c0bda6e5a1487fcd467&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dimx708_links.txt&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2119 (2.1K) [application/octet-stream]
Saving to: ‘imx708_links.txt’

imx708_links.txt              100%[=================================================>]   2.07K  --.-KB/s    in 0.001s

2024-01-04 05:35:52 (1.57 MB/s) - ‘imx708_links.txt’ saved [2119/2119]

[... 31 lines cut ...]
An unknown error occurred while installing dtoverlays.
dpkg: error processing package arducam-nvidia-l4t-kernel (--install):
 installed arducam-nvidia-l4t-kernel package post-installation script subprocess returned error exit status 255
Errors were encountered while processing:
 arducam-nvidia-l4t-kernel

Unknown error, please send the error message to support@arducam.com
jetson@jetson-desktop:~$ cd jetsonUtilities
jetson@jetson-desktop:~/jetsonUtilities$ python jetsonInfo.py
NVIDIA NVIDIA Jetson Nano Developer Kit
 L4T 32.7.3 [ JetPack 4.6.3 ]
   Ubuntu 18.04.6 LTS
   Kernel Version: 4.9.299-tegra
 CUDA 10.2.300
   CUDA Architecture: 5.3
 OpenCV version: 4.1.1
   OpenCV Cuda: NO
 CUDNN: 8.2.1.32
 TensorRT: 8.2.1.9
 Vision Works: 1.6.0.501
 VPI: 1.2.3
 Vulcan: 1.2.70
```

What am i doing wrong

Dion4cen, 2024-01-05:
Please stay tuned with this error. 
There is something unexpected occuring. We are checking the driver source code and will have it fixed.

edward-ardu, 2024-01-06:
@Thesis123123 

Thanks for submitting the bug. I have only adapted the 2gb version in nano, and I will update the 4gb version.

Thesis123123, 2024-01-07:
@Dion4cen @edward-ardu Thank you for the help, greatly appreciated. Hope to hear a solution soon

edward-ardu, 2024-01-08:
@Thesis123123 

I have fixed the issue, please re-execute the tutorial

Thesis123123, 2024-01-08:
@edward-ardu It installed with no errors this time, but im not able to open up the camera with the example commands. Am i using it right?

```
jetson@jetson-desktop:~$ v4l2-ctl --list-formats-ext
bash: v4l2-ctl: command not found
jetson@jetson-desktop:~$ gst-launch-1.0 v4l2src device=/dev/video0 ! video/x-raw, format=GAY8, width=2560, height=800 ! videoconvert ! xvimagesink
WARNING: erroneous pipeline: could not link v4l2src0 to videoconvert0, neither element can handle caps video/x-raw, format=(string)GAY8, width=(int)2560, height=(int)800
jetson@jetson-desktop:~$ cd jetsonUtilities
jetson@jetson-desktop:~/jetsonUtilities$ python jetsonInfo.py 
NVIDIA NVIDIA Jetson Nano Developer Kit
 L4T 32.7.3 [ JetPack 4.6.3 ]
   Ubuntu 18.04.6 LTS
   Kernel Version: 4.9.299-tegra
 CUDA 10.2.300
   CUDA Architecture: 5.3
 OpenCV version: 4.1.1
   OpenCV Cuda: NO
 CUDNN: 8.2.1.32
 TensorRT: 8.2.1.9
 Vision Works: 1.6.0.501
 VPI: 1.2.3
 Vulcan: 1.2.70
jetson@jetson-desktop:~/jetsonUtilities$ 
```

Here is the installation terminal of the driver
```
jetson@jetson-desktop:~$ wget https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/install_full.sh
--2024-01-04 06:47:10--  https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/install_full.sh
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/ebdc0c54-a520-42e6-b4ef-e241b3acfd59?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240108%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240108T070829Z&X-Amz-Expires=300&X-Amz-Signature=8d4ba860de426ce247e586273f776dce9e8c85ba131d04528b8f35379d9a4bf2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dinstall_full.sh&response-content-type=application%2Foctet-stream [following]
--2024-01-08 15:08:29--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/ebdc0c54-a520-42e6-b4ef-e241b3acfd59?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240108%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240108T070829Z&X-Amz-Expires=300&X-Amz-Signature=8d4ba860de426ce247e586273f776dce9e8c85ba131d04528b8f35379d9a4bf2&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dinstall_full.sh&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4560 (4.5K) [application/octet-stream]
Saving to: ‘install_full.sh.1’

install_full.sh.1             100%[=================================================>]   4.45K  --.-KB/s    in 0.001s

2024-01-08 15:08:29 (8.22 MB/s) - ‘install_full.sh.1’ saved [4560/4560]

jetson@jetson-desktop:~$ chmod +x install_full.sh
jetson@jetson-desktop:~$ ./install_full.sh -m imx708
./install_full.sh: line 19: warning: command substitution: ignored null byte in input
NVIDIA Jetson Nano Developer Kit
--2024-01-08 15:08:41--  https://github.com/ArduCAM/MIPI_Camera/releases/download/v0.0.3/imx708_links.txt
Resolving github.com (github.com)... 20.205.243.166
Connecting to github.com (github.com)|20.205.243.166|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/75c20efb-ee07-40e6-afe2-707089d55075?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240108%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240108T070842Z&X-Amz-Expires=300&X-Amz-Signature=94f07d9ba414ddc87b80766107184b0b83682d0189f212e2c5c4e9cde1b9b331&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dimx708_links.txt&response-content-type=application%2Foctet-stream [following]
--2024-01-08 15:08:42--  https://objects.githubusercontent.com/github-production-release-asset-2e65be/186627215/75c20efb-ee07-40e6-afe2-707089d55075?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAVCODYLSA53PQK4ZA%2F20240108%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20240108T070842Z&X-Amz-Expires=300&X-Amz-Signature=94f07d9ba414ddc87b80766107184b0b83682d0189f212e2c5c4e9cde1b9b331&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=186627215&response-content-disposition=attachment%3B%20filename%3Dimx708_links.txt&response-content-type=application%2Foctet-stream
Resolving objects.githubusercontent.com (objects.githubusercontent.com)... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to objects.githubusercontent.com (objects.githubusercontent.com)|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2119 (2.1K) [application/octet-stream]
Saving to: ‘imx708_links.txt’

imx708_links.txt              100%[=================================================>]   2.07K  --.-KB/s    in 0.02s

2024-01-08 15:08:43 (114 KB/s) - ‘imx708_links.txt’ saved [2119/2119]

[... cut ...]
(Reading database ... 181372 files and directories currently installed.)
Preparing to unpack arducam-nvidia-l4t-kernel-t210-4.9.299-tegra-32.7.3-20240108135140_arm64_imx708.deb ...
Unpacking arducam-nvidia-l4t-kernel (4.9.299-tegra-32.7.3-20240108135140) over (4.9.299-tegra-32.7.3-20230906151208) ...
ls: cannot access 'nv_imx477.ko.bak': No such file or directory
Setting up arducam-nvidia-l4t-kernel (4.9.299-tegra-32.7.3-20240108135140) ...
ls: cannot access 'nv_imx477.ko': No such file or directory

reboot now?(y/n):
y
reboot
```

edward-ardu, 2024-01-09:
@Thesis123123 

`sudo apt install v4l-utils`

Thesis123123, 2024-01-09:
> @Thesis123123
> 
> `sudo apt install v4l-utils`

@edward-ardu  I tried this but, i still get the erroneus pipeline error

[image]

Could there be something wrong with the example guide's pipeline? Apologies but im familiar with how to create gstreamer pipelines and have to relly heavily on guides. 

Do you know of the pipeline for taking an image with a specific resolution at a specific focus (this is what i really need) ?

edward-ardu, 2024-01-10:
@Thesis123123 

Please refer to the following how to use Gstreamer, and pay attention to modifying the resolution, frame rate and format.
https://github.com/ArduCAM/MIPI_Camera/tree/master/Jetson/IMX477/driver

Thesis123123, 2024-01-10:
@edward-ardu Can i control the focus using this driver? Gain, Exposure, Framerate, and Group Hold are the only one listed.

It may be of note none of the the example pipeline are working and are throwing the same error: "Erroneous pipelinen". But i was able to make it display something using this:

`gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! "video/x-raw(memory:NVMM),width=4608,height=2592,framerate=14/1" ! queue ! nvegltransform ! nveglglessink`

But i still dont have manual focus control, the autofocus is not even working. Is there anyway i can control the focus of the IMX708?

edward-ardu, 2024-01-10:
@Thesis123123 

> Can i control the focus using this driver?
There is currently no relevant driver control motor, but you can try using our script.
https://github.com/ArduCAM/Jetson_IMX708_Focus_Example/tree/main

Thesis123123, 2024-01-10:
> @Thesis123123
> 
> > Can i control the focus using this driver?
> 
> There is currently no relevant driver control motor, but you can try using our script. https://github.com/ArduCAM/Jetson_IMX708_Focus_Example/tree/main

This is great, I got everything i need now! My only complaints are why didnt i see this link on your online guides :<

[image]

closed 2024-01-10 by Thesis123123