ArduCAM/MIPI_Camera #189: Focus control on the IMX708 12MP Camera on the jetson nano

Thesis123123, 2024-01-10:
How do i control the focus for the IMX708, i would like to adjust the focus to a fixed point. Currently even the autofocus is not working.

Example G-streamer pipelines or python code examples for controlling the focus is greatly appreciated

edward-ardu, 2024-01-11:
@Thesis123123 
Bro, could you tell me more about your current situation? I saw in your last issue that everything is normal.

Thesis123123, 2024-01-11:
> @Thesis123123 Bro, could you tell me more about your current situation? I saw in your last issue that everything is normal.

Apologies, I thought it was off topic in the other question to ask about focus, so i made another thread. So far everything is working. will close this topic

But for those who might happen to stumble here basically focus is done using i2c and not the driver: Try out their script

>> Can i control the focus using this driver?
>
>There is currently no relevant driver control motor, but you can try using our script.
>https://github.com/ArduCAM/Jetson_IMX708_Focus_Example/tree/main
>
>_Originally posted by @edward-ardu in https://github.com/ArduCAM/MIPI_Camera/issues/188#issuecomment-1884142149_

closed 2024-01-11 by Thesis123123