ArduCAM/MIPI_Camera #191: IM708 display colors are wrong on the Jetson nano

Thesis123123, 2024-01-15:
I have installed the driver using the driver as per the [Quick Start guide](https://docs.arducam.com/Nvidia-Jetson-Camera/Native-Camera/Quick-Start-Guide/)

but the video display Image color seems to be off calibrated or has this greyish - red tint on it

[image]

I tried launching it with this command:

`gst-launch-1.0 nvarguscamerasrc sensor-id=0 ! "video/x-raw(memory:NVMM),width=4608,height=2592,framerate=14/1" ! queue ! nvegltransform ! nveglglessink`

and even tried adjusting the gain and exposure:

```
v4l2-ctl -c gain=12
v4l2-ctl -c exposure=6000
```

But i could get it to be capture decent colors. Any idea what could be wrong with it?

edward-ardu, 2024-01-24:
use
`sudo rm /var/nvidia/nvcam/settings/camera_overrides.isp
`
And then try command again

edward-ardu, 2024-01-24:
The Nano effect I used:
[image]

snnzdmr, 2024-10-23:
Hello, I have the same issue, but I don't have such a file in this directory. ( /var/nvidia/nvcam/settings/camera_overrides.isp)
I'm getting images from the camera, but the image (video) color is greyish.
sinan@sinan-desktop:/var/nvidia/nvcam/settings$ ls
nvcam_cache_0.bin  README.txt  serial_no_0.bin