ArduCAM/MIPI_Camera #192: AGX Orin can't detect IMX477 even after installing the driver.

NaJiwoong, 2024-01-23:
Hi there,
I'm using Jetson AGX Orin with Jetpack 5.1.1, and 5.1.2
I purchased the SKU:B0473 from arducam, and installed the driver in release.
I installed the driver using command "./install_full -m imx477".
Here, the driver isn't detected, but seldomly when I check the booting message with "sudo dmesg | grep -E "imx477|imx219|arducam".
(I can only see "arducam").
Additionally, sometimes the driver is detected, but still it doesn't detect the camera, which means "ls /dev/video*" cannot find anything.
Since I connected the camera and the board I received from Arducam, I think it has to works. (I followed the quick start guide, and checked the troubleshoot)
Quickstart: https://docs.arducam.com/Nvidia-Jetson-Camera/Nvidia-Jetson-Orin-Series/NVIDIA-Jetson-AGX-Orin/Quick-Start-Guide/
Troubleshoot: https://docs.arducam.com/Nvidia-Jetson-Camera/Native-Camera/Toubleshoot/
Could you check the driver deb file if it has any problem or not?
Thank you in advance.

Dion4cen, 2024-01-24:
Hi,
Let's contact with each other in the email. We need to check your device remotely.

missxa, 2024-07-16:
hi @Dion4cen, we're experiencing similar issues. the cameras are detected in 1 out 5 times after the driver are re-installed and jetson orin rebooted. we using B0473 multi-camera adapter as well. kernel is 
```
$ cat /etc/nv_tegra_release 
# R35 (release), REVISION: 4.1, GCID: 33958178, BOARD: t186ref, EABI: aarch64, DATE: Tue Aug  1 19:57:35 UTC 2023
```
```
$ uname -r
5.10.120-tegra
```
```
$ sudo dmesg | grep -E ardu
[sudo] password for alice: 
[    0.005153] DTS File Name: /home/jenkins/agent/workspace/x_orin_kernel_l4t-35.4.1-arducam/kernel/kernel-5.10/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t23x/concord/kernel-dts/tegra234-p3701-0005-p3737-0000.dts
```

Dion4cen, 2024-07-22:
Hi Alona,

This hardware version does have similar issues. We have upgraded the
hardware and you can contact the market for replacement.

You can contact ***@***.*** for the further steps.